EdgeTX/edgetx #5643: FrSKy Horus X10E lost hardware setup during flight?

BompaBert, 2024-10-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Lost control of the airplane during flight - receive still ok.. Radio pwr off/on showed loss of throttle calibration and pot / slider definitions. Pictures and log file attached - log, as from line 344, values change and freeze.
[LOGS.zip](https://github.com/user-attachments/files/17534653/LOGS.zip)

### Expected Behavior

Normal flight with airplane under full control.

### Steps To Reproduce

Unable to reproduce.

### Version

2.10.5

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

3djc, 2024-10-29:
That means SD card issue likely

BompaBert, 2024-10-29:
SD card check does not reveal an issue.

C:\Users\beprn>chkdsk G: /f /r /x
The type of the file system is FAT32.
Volume X10 SDCARD created 9/5/2024 2:36 PM
Volume Serial Number is B294-35BE
Windows is verifying files and folders...
File and folder verification is complete.
Windows is verifying free space...
Free space verification is complete.

Windows has scanned the file system and found no problems.
No further action is required.
    7,753,728 KB total disk space.
           32 KB in 1 hidden files.
        2,336 KB in 72 folders.
      147,552 KB in 963 files.
    7,603,776 KB are available.

       32,768 bytes in each allocation unit.
      242,304 total allocation units on disk.
      237,618 allocation units available on disk.

C:\Users\beprn>dir G:
 Volume in drive G is X10 SDCARD
 Volume Serial Number is B294-35BE

 Directory of G:\

09/05/2024  02:31 PM    <DIR>          RADIO
09/05/2024  02:31 PM    <DIR>          SCREENSHOTS
09/05/2024  02:31 PM    <DIR>          SCRIPTS
09/05/2024  02:31 PM    <DIR>          SOUNDS
09/05/2024  02:31 PM    <DIR>          TEMPLATES
09/05/2024  02:31 PM    <DIR>          THEMES
09/05/2024  02:31 PM    <DIR>          WIDGETS
08/04/2024  04:40 AM                 5 edgetx.sdcard.version
09/05/2024  02:31 PM    <DIR>          FIRMWARE
09/05/2024  02:31 PM    <DIR>          IMAGES
09/05/2024  02:31 PM    <DIR>          LOGS
09/05/2024  02:31 PM    <DIR>          MODELS
               1 File(s)              5 bytes
              11 Dir(s)   7,786,266,624 bytes free

C:\Users\beprn>

pfeerick, 2024-10-30:
> Radio pwr off/on showed loss of throttle calibration and pot / slider definitions. 

Can you be more specific here - when you restarted the radio, did it *prompt* you to calibrate, or was the calibration off?

Did you get the "Unable to read valid radio settings" error on startup? As if that is the case, the loss of calibration is expected, and you should have a `radio_error.yml` file in your RADIO folder in addition to the current `radio.yml`. 

I would normally have thought this was an SD card issue, but that seems somewhat implausible here since SD logging continued for a good 30 seconds afterwards... 🤔

BompaBert, 2024-10-30:
After power up, calibration was off so radio did not prompt to calibrate however a throttle warning occurred: Throttle warning - not idle.
[image]
 No further errors - no radio_error.yml file.