EdgeTX/edgetx #5646: SBUS trainer lags after a while (SBUS-inverted signal from ER-6 onto TX16S)

dirteat, 2024-11-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have set up a trainer mode to connect 2 TX16S emitter. For this I am using:

- An ER-6 receiver, configured with SBUS inverted serial, 100Hz, and plugged onto the TX16S_Teacher with the TX port of the ER-6 connected to the RX port of the TX16S_Teacher on AUX1
- AUX1 on the TX16S_Teacher is set as "SBUS Trainer" and powered.
- The TX16S_Learner binds to the ER-6 receiver

Switching trainer mode on the TX16S_Teacher, I am receiving the input of the TX16S_Learner, first without issues (Chan or Stick). But after a while, tens of seconds, maybe a minutes, the input start to lag, crazily lags (would be catastrophic while flying).

I've tried two different ER-6 receivers, same symptoms. What seems to not lead to catastrophic lag is to set the Packet Rate to 333Hz (full) on both the TX16_Learner and the ER-6!?? But in that situation, the controlled surfaces on the plane as ordered by the TX16S_Learner through the trainer TX16_Teacher do not move really smoothly, they seem to jump by little steps.

Let me know if you want me to do other tests. But, for the time being, I am not going to teach with the setup :)

NB: All these guys are running exactly the same firmwares and softwares, namely: edgeTX 2.10.5, expresslrs 3.5.1

### Expected Behavior

I would expect the input to remain steadily smooth at all times no?

### Steps To Reproduce

See description above.

### Version

2.10.5

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-11-04:
Cannot reproduce. I have it running for over an hour, still silky smooth.

dirteat, 2024-11-04:
Damned. Same hardware, same software, same receiver?

3djc, 2024-11-04:
Roughly yes, I'm just on elrs 3.5.0. Where is your lag occurring? How is the output widget doing ? Is is jerky there too ?

dirteat, 2024-11-04:
Yes, the widget output, then showing the input from TX_learner, is visibly laggy as well!

There is only one thing I have not tested twice, the cable connecting the ER-6 to the TX_teacher (I've checked with a spare ER-6, but I used the same cable to connect it to AUX1). I'll test that and report...

mha1, 2024-11-04:
Try these setting:
- 333Hz on both radios
- 10mW fixed and telemetry off on the student radio

philmoz, 2024-11-04:
I tried it as well and could not reproduce the issue (TX16S for master, T20V2 for slave).

Can you post all of the settings for the ER6:
- Internal / External RF settings on the slave radio
- ELRS config lua script settings on the slave radio

3djc, 2024-11-05:
Possibly unrelated too, but turning off telemetry on subs inverted receiver is likely a good thing to do in general, especially if somehow fitted in case or close to radio main board

55Laurent, 2024-11-05:
ok, thanks, I'll try

3djc, 2024-11-05:
That is more for interactions between the 'trainee' receiver and the master radio in very close proximity

55Laurent, 2024-11-05:
understood, this why I set the telem power to the min value, and I found really no issue, whatever the positions of the 2 tx (I installed the buddy receiver is inside the master TX16S). But I'll try with buddy telem off as you said.
Then the only issue I got is when there is no model match number, the binding between pupil tx and buddy receiver is ok, but there are very big lags and in take almost 1mn to disapear. All firmwares are up to date.

Gerold68, 2024-11-05:
I opened an issue 2 years ago for this strange behavior. Seems it was never solved completely.

3djc, 2024-11-05:
Understood @55Laurent that was just a generic and general comment. Can't seem to replicate, but as weird as it sounds, it does indeed sound elrs given what you say

dirteat, 2024-11-08:
All thanks for all your input.

I have changed the cable connecting the ER6 to the TX_trainer, and I still do observe the lagging getting in after a while. This times, I went on playing the commands even though they were lagging, and the lag disappears after another while, to come back later on.

Importantly, this happens only at 100Hz (full) between TX_Learner and ER6: At 333Hz (full), this is not happening.  I have also tried various power output for the TX_Trainer and TX_Learner, as suggested, and the power output does not change anything.

Also, this "seems" correlated to a loss of signal as, indeed, I have telemetry on between TX_Trainer and ER-6, precisely to monitor the radiolink. When "telemetry loss" pops out, then, very likely the lagging starts a few seconds afterwards ( not 100% of the time though).

NB: I have made a video showing the lag, not sure if I can post it there, but I'll try in another comment.

dirteat, 2024-11-08:
Ok, the video is too big (60MB), here a link to my personal website:

[sbus_lagtest.webm](https://curl.irmp.ucl.ac.be/~chris/upload/aero/sbus_lagtest.webm)

The video starts while the lag was already in, I am toying for a while showing it. Then, I decide to power off and on the ER-6, the bind TX_learner ER6 is auto-magically restored, but the lag is still there, and, finally disappear after ten more seconds or so.
I insist that the lag can also disappear without resetting the ER-6.

3djc, 2024-11-09:
I have checked the signal getting out of ER6 when trainer radio is set to 100Hz or 333Hz. There is a small period difference (10ms vs 7ms), but nothing that could disrupt ETX.

There is a question on how on earth you can have "telemetry lost" between trainee radio and the ER-6. That should never happen at that range.

Also the fact that ER-6 rebinds to master tx absolutely makes no sense. At this point in time, if problem isn't an issues with your settings, I think it point toward ELRS issues, not ETX

dirteat, 2024-11-09:
```There is a question on how on earth you can have "telemetry lost" between trainee radio and the ER-6. That should never happen at that range```

It is actually something I observed even without doing this setup, between my TX and the plane, when both are very close... In flight, this never happens. Anyway, yes, that points more to ELRS than EdgeTX indeed. I should probably report the issue there :-/

mha1, 2024-11-13:
@3djc I was able to reproduce this and checked the ELRS side. SBUS data delivered at AUX1 looks good and smooth while the problem can be observed looking at a jerky lagging channel monitor on the master radio. See my comment at the twin ELRS issue.

To me it now looks like a sampling problem.

mha1, 2024-11-14:
@3djc to support the timing theory some side notes. ELRS attempts to output SBUS frames at a 9ms period, 50Hz and 100Hz packet rates won't be able to achieve this as fresh channel data can only be delivered at 20ms (50Hz) and 10ms (100Hz). Selecting 100Hz packet rate will yield a 10ms period. This is exactly the mixer task period +- some minor jitter. For testing I fixed the SBUS period at 10ms. EdgeTX responded to this with random and sometimes longer periods of "trainer lost" while I was still observing SBUS frames arriving at AUX1. My suspicion is if the SBUS frame rate matches the mixer task period very accurately the SBUS data acquisition fails. The incoming SBUS jitter makes it look random. Fixing the period at 10ms made it worse by matching the mixer period more often. I can imagine a situation where every time the mixer tasks tries to fetch SBUS data an incoming SBUS frame is just processed meaning lost for this cycle. Double buffering should solve this. This would also explain why slave radio packet rates >100Hz, e.g. 333Hz wont show the problem, but using 333Hz doesn't fix the problem, it just masks it. Hope that helps.

gagarinlg, 2024-11-14:
Maybe we should to what FrSky did and start a clean sheet implementation of EdgeTX for Version 4

mha1, 2024-11-14:
> Maybe we should to what FrSky did and start a clean sheet implementation of EdgeTX for Version 4

Quite the radical fix for this issue 😜

3djc, 2024-11-15:
Indeed a bit radical ! It is difficult currently for me to work on this, but i want to try to understand if the fail is in the capture itself or decoding of the sbus stream. Out of curiosity, at what baudrate is your internal module set to ?

mha1, 2024-11-15:
both slave and master radios are on 921k (my go to setting)

3djc, 2024-11-17:
@mha1 could you test with this binary (beware, it is based on main, not 2.10)

mha1, 2024-11-17:
> @mha1 could you test with this binary (beware, it is based on main, not 2.10). I don't seem to be able to reproduce with it
> 
> [firmware.bin.zip](https://github.com/user-attachments/files/17790698/firmware.bin.zip)

Raphael's change? Looking good. I tested various packet rate combinations on the salve and mnaster radio (even slave radio at 50Hz and F1000) and I also tested this with a fixed 10ms SBUS rate vs 100Hz packet rate on the master radio (SBUS period = Mixer period) which previously resulted in a lot of trainer lost messages. All looking good and I have the impression it's more lively and smoother than it ever was. Wolud be nice if  @dirteat could confirm this in his environment.

3djc, 2024-11-17:
Yes it is smoother as we read as much as available (on serial idle)

mha1, 2024-11-17:
Does it fit 2.10 too?

pfeerick, 2024-11-24:
It will cherry-pick in with some irrelevant cosmetic changes (i.e. include guard, opentx-edgetx include changes), and still compiles for the TX16S, but whether it will works properly or not I'm not sure yet. But it needs to be in nightly and tested before it can be considered for 2.10.x ;)

3djc, 2024-11-24:
I'm not too sure it is worth the risk of adding it to 2.10. Granted, it fixes a true issue, but also, you can work very simply around the issue by selecting an ELRS frame rate other than 100Hz on the master radio

closed 2024-11-25 by pfeerick

dirteat, 2024-11-26:
> if @dirteat could confirm this in his environment

I'll try, I got recently initiated to STM32CubeProg, so I am less afraid to be bold now ;)

PR #5672 (2024-11-25): feat: trainer SBUS processing in IDLE IRQ
Fixes #5646

Summary of changes:
Read SBUS serial on aux idle irq (so as soon as pulse as finished) instead of periodic 5ms . Only complete checked frames will update trainer data.

Also completes a TODO ! Code by Raph and Malte