EdgeTX/edgetx #5650: companion as a portable install in Windows

alfredborges, 2024-11-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Different firmware versions require different companion versions to manage in Windows.  So in Windows you can have two or more versions of companion installed on every PC you use to manage edgetx radios

### Describe the solution you'd like

I would be great to have the option of a "portable" install , so you could manage several firmware versions with an install.    This appears to be the behavior in the Linux version

### Describe alternatives you've considered

1-Run the Linux version
2- Go ahead and install multiple versions of companion

elecpower, 2024-11-05:
When you run the Windows installer add the version to the install folder name. You can then create Desktop shortcuts to each version.

alfredborges, 2024-11-05:
That's a good workaround for our current condition

I was looking to avoid multiple installs and have have the multiple instances self contained

pfeerick, 2024-11-05:
Might need to be a little more specific as to use case. As individual minor
version installs... i.e. 2.8.x, 2.9.x are effectively self contained, as
they don't even share Companion settings. You probably just can't, for
instance copy the install folder to a flash drive and in doing so copy the
program AND settings in one near bundle. Could probably do something using
the PortableApps launcher packager though.

alfredborges, 2024-11-06:
Not clear as to your point

I try to have a folder for each firmware revision that I've installed on a radio
For my Zorro , I have folders for 2,70 (stock) , edgetx 2.81,294 and 2.105.  I need separate companion instances for these separate firmware folders

pfeerick, 2024-11-06:
My point was/is that you merely have to install them to separate folders, and since they each use their own major and minor version number when saving settings in the registry (i.e. 2.9, 2.10) they are effectively isolated from one another. 

```
HKEY_CURRENT_USER\Software\EdgeTX
```

Or, it is possible to wrap stuff using the PortableApps.com launcher to make an app AND settings self-contained... (within reason - some apps will still need some external dependencies installed). 

i.e. if you trust some random stranger on the internet... below is a self-extracting package created using the [PortableApps.com](https://portableapps.com/) launcher and installer toolkits... First time you run the app, it will create a Data folder in that same folder, and keep a private copy of the registry settings it loads into memory, and removes from memory every time you run that app. Configuration for the launcher is in the App\AppInfo\AppInfo.ini and App\AppInfo\Launcher\CompanionPortable.ini files. Would only need a couple of minor tweaks (basically just the version numbers changed) so that the launcher was correct for the different version of Companion being bundled. 

The zip file is the same content as the self extractor, just not an executable, if you prefer to check things out first ;) 

Self-extracting package - http://files.portabledev.xyz/f/0564eb9b0b964442b969/
Zip file - http://files.portabledev.xyz/f/6a6f3c5739834469a988/