EdgeTX/edgetx #5651: Simulator with network Server SD Structure path need minutes to open

elekSENDER, 2024-11-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

After selecting a SD Structure path folder with a server adress (i.e: //server01/public/EdgeTX/SD_TX16S) everything in companion looks normal and functionable. SD synchronizing is working, models can be opened and were editable.
Until opening simulator: startup screen is coming up, but then the simulator holds on startupscreen for several minutes (3-5min). After this time the simulator is coming up und works normally.
At this "freezetime" on the taskmanager ethernet network monitor occurs a weird traffic with low usage (approx. 1-2Mbit/s up and down on a 1Gbit connection) from opening the simulator till normal working.
By closing the simulator window in freeszetime, companion freezes and can only be closed by taskmanager or with the window which tells you companion.exe does not work any more (close/wait for programm).
If you wait after closing simulator until the slow ethernet traffic was over companion is working well again.

This was a former issue in OpenTX as well, in the later versions (I tried OpenTX 2.3.15) the simulator works fine with a network adress.

### Expected Behavior

Simulator normal working with network path.

### Steps To Reproduce

Enter/choose a network path in SD Structure path.
Synchronize the SD Card
Open any model.
Start simulator.

### Version

2.10.5

### Transmitter

FrSky X10 / X10S (ACCST), RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 10 home ver22H2

elekSENDER, 2024-11-06:
same behavior on a Windows 10 Pro ver 22H2

elekSENDER, 2024-11-06:
and the same on a Windows 11 Home ver 23H2

elecpower, 2024-11-06:
Test model TX16S with one special function ON Play Track 4wd repeat 3s.

EdgeTX 2.10.5 with TX16S sim on Win 10 22H2 with SD path set to:
- C:
- mapped drive to Open Media Vault SMB share; and
- unc path to Open Media Vault SMB share

Results: no noticeable difference

So it would appear it relates to the model settings eg widgets or its your IT environment.

elecpower, 2025-01-04:
@elekSENDER Have you had any luck with improving performance?

elekSENDER, 2025-01-05:
no, not really improvments, but as I mentioned above in an openTX-companion this extremly latency do not occur and I did some testing with a .otx file with conversion to a .etx file an so on.
Due to that I am Austrian and my motherlanguage is german I like to read the dispaly in english but hear the sounds and alerts in german. So i have only installed the DE-sounds and while testing suddenly the simulator starts in the normal time!

Long boring story, short message: ;-)
If you choose a language behind the button "Edit Radio Settings" in the .etx file which ist NOT installed in the SD-structure path the simulator starts normaly. Of course you can hear no sounds.
And vice versa: any installed sound directory (i tried DE and EN) correspondig to the radio settings causes the long waiting time with a very slow traffic on the ethernet port, but then sounds are ok.

I fully understand that the data transfer rate to a NAS is much slower than to a SSD or HD, but maybe you can find something with this information.

pfeerick, 2025-02-07:
If you can check again with a recent build from main, or from a release candidate for 2.11 once they become available, that would be great, as there were some improvements made to file loading that should have helped with this.

elekSENDER, 2025-02-17:
Yes, I´ve tested 2.11.0-RC1 and there was a huge decrease in loading time in my - obviously - special environment.
It is not on the level of the old OpenTX, but tested with 2.11.0-RC1 I have approx. 25% of loading time (which is now in my case 52sec) in comparison to 2.10.6.
So this a really big improvement, thanks a lot for taking me serious!

What I can still see, is this weired writing traffic to the SD-path on the network monitor while waiting to the complete opening of the simulator.
And its still possible by choosing in "Edit Radio Settings" in the .etx-file a language which is not installed in the SD-path to reduce the opening time to 2sec.
So my guess is there is any connection with usage of the sound-files.

closed 2025-12-27 by elekSENDER