EdgeTX/edgetx #5652: Timer 3 disables when radio restart

Macianno, 2024-11-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I use a Timer 3 as a count-up its default value is 1:00:00 instead 00:00:00. Additionally it disable itself after radio is turned off - doesn't matter if count-up or count-down set.
I have TX16SMK2 transmiter with edgetx-tx16s firmware, v2.10.4"Centurion".

### Expected Behavior

Timer 3 stay being enabled after transmitter restart and keep settings.

### Steps To Reproduce

1. Go into model menu
2. Enable Timer 3
3. Set Mode: ON
4. Set Persistent: Manual Reset
5. Preview Timer 3 value - it is 1:00:00
6. Turn off the radio
7. Turn the radio on back
8. Go into model menu - timer 3 is disabled

### Version

2.10.4

### Transmitter

RadioMaster TX16S / TX16SMK2

philmoz, 2024-11-08:
Not reproducible on my TX16S.

I would guess you have a Lua script somewhere that is messing with the timer.

Macianno, 2024-11-08:
I just removed INAV telemetry script running on additional screen and Timer 3 doesn't reset anymore. Thank you @philmoz :)

closed 2024-11-08 by Macianno