EdgeTX/edgetx #5656: Invalid coefficient of trim value

chofchop, 2024-11-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the current EdgeTX, the trim value takes values ​​from -1049 to 1049.
[image]

### Expected Behavior

The trim value should be -1024~1024.
 It has been multiplied by an extra 1.024 somewhere.

1024 x 1.024 = 1048.576 ≒ 1049

### Steps To Reproduce

Display the trim value using Lua below.

local valuesToTest = {
  'ste',
  'thr',
  'trim-ste',
  'trim-thr',
  'trim-t3',
  'trim-t4',
  'trim-t5'
}
local run = function(evt, ts)
  lcd.clear()
  for i, valueName in ipairs(valuesToTest) do
    local value = getValue(valueName)
    lcd.drawText(10, (i - 1) * 8, valueName..' : '..tostring(value))
  end
  return (evt == EVT_EXIT_BREAK) and 1 or 0
end
return { run=run }

### Version

2.10.5

### Transmitter

RadioMaster MT12

philmoz, 2024-11-10:
It has been like this for a long time - at least 2.8 probably a lot longer.

While it could be changed, there is a risk of affecting existing scripts.

chofchop, 2024-11-10:
Yes, I know that too, but that excuse is not reasonable.
If this obvious mistake is ignored, the number of affected scripts will continue to increase. I can't understand why they didn't fix it as soon as they noticed it.

pfeerick, 2024-11-10:
1) This may be the first time it has been noticed, since it is still present, no?

2) This looks like it is only a minor scaling issue, so the risk of affecting existing setups is minimal and at worst requires a minor adjustment. Better to ascertain the cause and resolve it sooner than later.

philmoz, 2024-11-10:
The cause is this line in mixer.cpp:
```C
    return calc1000toRESX((int16_t)8 * trim_value);
```

trim_value range is -128 to 128 (or -512 to 512 with extended trims).
8 * trim_value range is -1024 to 1024.

calc1000toRESX assumes the input value range is -1000 to 1000 and scales it up to -1024 to 1024.

chofchop, 2024-11-11:
I am testing by rewriting line 421 of mixer.cpp as shown below.

    return (int16_t)8 * trim_value;

Although I have only been able to confirm part of it, it seems to be working properly on both the OS and the script so far.

PR #5674 (2024-11-24): fix(radio): trim values scaled incorrectly when displayed via Lua
Remove incorrect scaling of trim values.

Fixes #5656
closed 2024-11-24 by pfeerick