EdgeTX/edgetx #5662: unable to get orange DSM2 Rx throttle channel to work. Works ok with Spektrum DX9

ror1948, 2024-11-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

- bind to Rx with DSM "2 1F" 22ms no problem
- Tx throttle set to LOW
- ESC  initializes with 3 beeps when using 3S LiPo.
- unable to get throttle channel low set to 995us(max throw and extended )
- measured cycle rate is 4.5ms when it should be 22ms
- 4in1 version 1.3.4.0 AETR
- other channels seem to work with reduced range

- 

### Expected Behavior

- bind to Rx with DSM "2 1F" 22ms 
- able to get throttle channel low set to 995us(max throw and extended )

### Steps To Reproduce

- bind to Rx with DSM "2 1F" 22ms 
- set max throw on
- set throttle chanel low to -120
- note that ESC does not initialize with 3 beeps when using 3S LiPo.
- 

### Version

2.10.5

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-11-13:
More an issue for https://github.com/pascallanger/DIY-Multiprotocol-TX-Module ?

ror1948, 2024-11-13:
moved to https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1044

thanks

closed 2024-11-13 by ror1948