EdgeTX/edgetx #5673: Fatfish-F16---EdgeTX-2.10.5-DE---Display-is-upside-down

udodill, 2024-11-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Fatfish F16 transmitter is delivered with EdgeTX version 2.10.0-selfbuild, date 2024-05-20. After updating EdgeTX to version 2.10.5, the output on the display is upside down.

### Expected Behavior

The information on the display should be displayed normally.

### Steps To Reproduce

Turn on the transmitter.

### Version

2.10.5

### Transmitter

Fatfish F16

### OS Version

not relevant

### Anything else?

[image]

richardclli, 2024-12-04:
There are 2 versions of Fatfish hardware, in which the LCD orientation is different, and the firmware should be able to detect and choose the correct orientation properly.  It is strange that the LCD orientation detection did not operate properly.  The detection is done by the PCBREV information hardwired in the hardware.  Not sure what is wrong though.

udodill, 2024-12-04:
Hello, Thank you very much for the feedback. In edgetx 2.11.0 Nightly the error is gone and the display is the right way round again.

Udo Dillschneider, Homburg / Saar

closed 2025-02-07 by pfeerick