EdgeTX/edgetx #5676: Trainer Mode Master/Serial turns Off when reading models and settings from radio to Companion

clarke-john, 2024-11-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When I ‘Read Models and Settings from Radio’ to Companion the setting for the ‘Trainer Mode’ changes from ‘Master/Serial’ to ‘Off’.
‘Write Models and Settings to Radio’ from Companion works fine and the Trainer Mode setting is not lost in this transfer.
I noticed this bug in February (version 2.9.4) but didn't report it.

### Expected Behavior

I would expect the Trainer Mode setting on the radio to be transferred to Companion.

### Steps To Reproduce

On a model in the radio, set 'Trainer Mode’ to ‘Master/Serial’.
Open Companion and connect radio to PC.
In Companion click on ‘Read Models and Settings from Radio’.
Open the model.
Check the setting for the ‘Trainer Mode’ and note that it has changed from ‘Master/Serial’ to ‘Off’.

### Version

2.10.5

### Transmitter

FrSky X10 / X10S (ACCST)

### Operating System (OS)

Windows

### OS Version

W11

pfeerick, 2024-11-24:
I am able to reproduce this on TX16S also. 
- Set AUX1 to "SBUS Trainer", to enable the "Master/Serial" trainer option. 
- Configure Trainer Modeto be Master/Serial on handset
- Companion reads as OFF, although "Master/Serial" is in the list of options.

Also, "Master/SBUS Module" is missing as a trainer option in Companion, and results in a blank (not OFF) field.

elecpower, 2024-11-24:
Thanks @pfeerick more work lol

pfeerick, 2024-11-24:
Yeah, I know... that todo list never gets any shorter! 😆

pfeerick, 2024-11-24:
Hm... #4266 is at least part of the reason for this, as it dynamically checks if the radio has hardware capabilities (which allowed a few more radios to have SBUS/CPPM option and also checks if it is in use by something else...

closed 2024-11-29 by pfeerick

clarke-john, 2024-11-30:
Thank you all very much for fixing this bug.  I look forward to it in the next release.

 

I also appreciate all the work and dedication the devs put into giving us this superb firmware for our radios.

 

Best wishes

 

John

 

Closed #5676 <https://github.com/EdgeTX/edgetx/issues/5676>  as completed via #5684 <https://github.com/EdgeTX/edgetx/pull/5684> .

pfeerick, 2025-01-29:
Oh fiddlesticks, I think I forgot to include that fix in 2.10... it looks like it is only in for 2.11 :-/

clarke-john, 2025-01-29:
Thanks Peter, I really appreciate what you guys do for us.  It is a brilliant system.

pfeerick, 2025-01-29:
You'll be able to download a 2.10 Companion build with this fixed shortly:
- Linux - https://github.com/EdgeTX/edgetx/actions/runs/13042188324
- MacOS - https://github.com/EdgeTX/edgetx/actions/runs/13042188303
- Windows - https://github.com/EdgeTX/edgetx/actions/runs/13042188300

PR #5684 (2024-11-29): fix(cpn): some trainer options not working
Fixes #5676

Summary of changes:
- allow Master/SBUS Module selection for more radios
- use correct enum for Master/Serial