EdgeTX/edgetx #5680: LVGL line should supported dashed lines

arunh, 2024-11-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The new LVGL Lua API coming in 2.11 does not support a way to draw dashed lines. This is a useful graphical device and used by a number of widgets, and makes it hard to port a widget to the new library if it needs this functionality.

### Describe the solution you'd like

The LVGL library itself has support for dotted and dashed lines so this just needs to be exposed in the EdgeTx Lua API.

https://docs.lvgl.io/8/overview/style-props.html#line-dash-width
https://docs.lvgl.io/8/overview/style-props.html#line-dash-gap

philmoz, 2024-11-22:
In LVGL dash gap and dash width only work for horizontal and vertical lines.

If that limitation is acceptable I can add these as properties to the 'hline' and 'vline' types.

arunh, 2024-11-22:
Horizontal and vertical is acceptable. Thank you.

PR #5681 (2024-11-23): feat(Lua): add properties for dashed vertical and horizontal lines in Lvgl API.
Closes #5680

Add 'dashGap' and 'dashWidth' properties to the 'hline' and 'vline' objects.
closed 2024-11-23 by pfeerick