EdgeTX/edgetx #5683: Pots & Slders

swissman, 2024-11-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

S1 Pot is the only pot or slider that shows up on the calibration screen and on the model page on my Radio master TX16S, with Version v2.10.5 Centurion .
Updated to 2.10.5 from 2.10.4, as also not working on that version.

### Expected Behavior

See pots and sliders

### Steps To Reproduce

See Error

### Version

2.10.5

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Window 10

### Anything else?

[image]
[image]

pfeerick, 2024-11-24:
Go to Radio Settings/SYS -> Hardware, look for the Pots button near the
bottom under inputs heading... You probably need to configure them... i.e.
pot with detent, multipos switch, pot with detent, slider, slider. Leave
the toggles at the end of the line alone, they are for inversion.

swissman, 2024-11-25:
Hi Peter.Many thanks for your help with this .Regards Trevor.Sent from my iPad

closed 2024-11-25 by pfeerick