EdgeTX/edgetx #5702: GX12 RF2TLM LUA Script Blows Up UI

jmxp69, 2024-12-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

GX12 Bone Stock with one model configured.
Copy rf2tlm.lua to SCRIPTS/MIXES from https://github.com/rotorflight/rotorflight-lua-scripts/releases
Add rf2tlm in any CUSTOM SCRIPTS line (I tried several).
Discover RotorFlight sensors using ELRS Custom Telem from RotorFlight Configurator 2.1.0-RC1
All extended sensors appear. Looks to be working ok.

Exit Telemetry Screen with RTN. Main screen appears and works.
Press SYS  Button "No tools found"
Press MDL Button "No models listed" and cannot navigate to model settings
Reboot Radio
Press SYS Button all scripts present
Press MDL Button all models present and can navigate to model settings
Go to telemetry sensors, all extended sensors gone.
Go to custom scripts page, rf2tlm still listed.
Go to file manager -> scripts -> mixes -> delete luac
Re-add rf2tlm to custom scripts line
Discover sensors
All extended sensors show
Exit telem
Press SYS Button "No tools found"
Press MDL Button "No models listed"
Rinse/Repeat

FW: edgetx-gx12
VERS: 2.11.0-selfbuild
GIT#: d5552c80

Also, RF2 LUA does not compile/run. 

rf2tlm.lua is confirmed working perfectly on TX16s with 2.10.5.

I have completely restored SD card to stock.
I have copied ONLY rf2tlm.lua to SCRIPTS/MIXES
Behavior repeats.

### Expected Behavior

rf2tlm should allow discovery of sensors and not impact normal operation of radio. i.e. Should not 'lose' tools and should not 'lose' model listing and should be able to navigate.

### Steps To Reproduce

1. Copy rf2tlm.lua to scripts/mixes
2. Add rf2tlm to custom scripts
3. Discover sensors using Enable Custom Telem on RF
4. Exit telem screen and navigate to basically anything other than the home screen

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

### Operating System (OS)

macOS

### OS Version

Sequoia 15.1.1

### Anything else?

FW: edgetx-gx12
VERS: 2.11.0-selfbuild < -Factory firmware and [GX12 SASD 04122024 v2.bin]
GIT#: d5552c80

3djc, 2024-12-11:
Could you try with this firmware, was able to reproduce with older, but no this one (so unsure if my reproduce process is wrong, or fixed in this version)
[TEST GX12.zip](https://github.com/user-attachments/files/18094354/TEST.GX12.zip)

jmxp69, 2024-12-11:
FW: edgetx-gx12
VERS: 2.11.0-selfbuild
GIT#: 13cee43b
DATE: 2024-12-11 11:56:19

Same exact behavior. Pattern repeats.

3djc, 2024-12-11:
Ok, need to understand why I cannot seem to replicate

jmxp69, 2024-12-11:
We could arrange a video demo if you want. Are you on my discord? https://discord.gg/jSFTASU or some other discord?

3djc, 2024-12-12:
Joined linked discord

3djc, 2024-12-13:
There seems to be a SPI bus crash involved here. Investigations in progress

3djc, 2024-12-15:
Because it provides more hardware than any other BW radio (switches, customizable switches, RGB leds,..), the memory available for LUA is a bit less than other radio. Unfortunately, this prevent Rotorflight LUA to properly compile, and that's what you are seing here.

A solution to run those script compiled under windows will be available when LUA 5.3 gets merged as part of 2.11. Ideally we should also be handling LUA out of memory situations better.

jmxp69, 2024-12-15:
Thanks for investigating. Is there anything the user (configuration wise) can turn off to reclaim enough memory for the script to work? i.e. Disabling features? This is a bit of a dealbreaker for RF users.

pfeerick, 2024-12-15:
For the present time no, probably not. lua53 will hopefully be merged this week, which brings with it memory savings amongst other things, and then a nightly for the GX12 can be used. We have a little time for that to stabilise for the GX12 since it still pre-order only.

jmxp69, 2024-12-16:
Peter/JC, I can confirm taking the scripts folder off the TX16s and placing on GX12: Both RF2 and rf2tlm appear to be working ok. I can run the RF2 tool to adjust RF settings and rf2tlm now survives reboot as my sensors stay present after rebooting. Glad there's a work around. I hope LUA53 allows more traditional install of these scripts.

3djc, 2024-12-16:
No, LUA5.3 won't help that much. it will allow you to compile the lua files on the simulator,  which might be needed (or not, depending on how the lua is written) for best compatibility with this radio model. But in any case, compiled on another radio or simu, the luac files will need to be included for this radio (and possibly others)

3djc, 2024-12-16:
> Thanks for investigating. Is there anything the user (configuration wise) can turn off to reclaim enough memory for the script to work? i.e. Disabling features? This is a bit of a dealbreaker for RF users.

Unfortunately not, there is no option for that. The only option (LUA5.3 or not) is to add the luac files

philmoz, 2024-12-17:
It doesn't appear to be a memory issue; but seems to be related to the SD card writing for the GX12.

PR 5716 adds a workaround that fixes this on my GX12.

jmxp69, 2024-12-17:
VERS: 2.11.0-selfbuild
GIT#: d7601437
DATE: 2024-12-17 01:47:20 

Appears to fix it. I remove all luac files. Reboot. Sensors remain. Tools-RF2 runs fine. rf2tlm remains in the custom scripts page as expected.

jmxp69, 2024-12-17:
I was able to get a blinking ! on the top left of the screen entering and exiting ELRS3 LUA from tools menu. When that appeared I couldn't navigate anywhere on the radio--had to power down.

jmxp69, 2024-12-17:
No, it only happened one time going in and out of ELRS lua.

philmoz, 2024-12-17:
> No, it only happened one time going in and out of ELRS lua.

Was it in the same run where you first ran the RF2 scripts to compile them?

jmxp69, 2024-12-17:
Yes, it was on the initial power up after clearing all compiled luas from sd card.

3djc, 2024-12-17:
> VERS: 2.11.0-selfbuild GIT#: [d760143](https://github.com/EdgeTX/edgetx/commit/d760143781032c1e3ff9c53ce425d5c7d432db6a) DATE: 2024-12-17 01:47:20
> 
> Appears to fix it. I remove all luac files. Reboot. Sensors remain. Tools-RF2 runs fine. rf2tlm remains in the custom scripts page as expected.

Said PR aim is not to fix the issue, it clearly doesn't. What it does/may do is prevent it from happening, but the underlying issue hasn't been identified, nor fixed, yet

jmxp69, 2024-12-17:
> Said PR aim is not to fix the issue, it clearly doesn't. What it does/may do is prevent it from happening, but the underlying issue hasn't been identified, nor fixed, yet

All I know is the uncompiled scripts don't bomb and cause loss of UI navigation ability. From the outside looking it, the issue I encountered is no longer occurring. If there's more work to be done to make it completely right, ok. But my gripe is resolved.

PR #5716 (2024-12-22): fix(radio): writing of .luac files can fail on some radios
When generating the compiled .luac file for a script the file write may be done with a large number of very small writes to the SD card. For some reason this is failing on the GX12 with some scripts.

This PR buffers the data being written when a .luac file is generated and writes it in larger blocks to the SD card.
This is a workaround for the GX12 issue; but may also reduce the SD write overhead when creating .luac files.

**Note: the underlying issue with the GX12 SD writing should still be investigated and fixed.**

Resolves #5702  [... cut]
closed 2024-12-22 by pfeerick