EdgeTX/edgetx #5704: UI overlay at radio shutdown

mha1, 2024-12-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

[image]

### Expected Behavior

Black background

### Steps To Reproduce

1. Power radio and connect to receiver
2. Edit SF and bring up trigger selection

[image]

3. Shutdown radio with receiver still connected

### Version

Nightly (Please give date/commit below)

### Transmitter

Other (Please specify below)

### Anything else?

T15 - current nightly t15-3d1e7a9.bin

philmoz, 2024-12-16:
There can be any content and number of layers present when a warning dialog like this pops up.
At the moment the code assumes a single layer and tries to hide it - which of course does not work in this case.

Simplest solution is to make the warning dialog background opaque:
[image]

pfeerick, 2024-12-16:
Something similar happened again the other day with the failsafe warning.
Full screen alert should always take exclusive control of the full
screen/view so opaque background would work. Wasn't there an image shown
there at some point?

philmoz, 2024-12-16:
Current warning will show the theme background if there is only one layer to hide.

closed 2024-12-16 by pfeerick