EdgeTX/edgetx #5708: On-screen widget will disappear after reboot if there is Chinese string in the widget lua file

L1cardo, 2024-12-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On-screen widget will disappear after reboot if there is Chinese string in the widget lua file.

### Expected Behavior

On-screen widget do not disappear after reboot if there is Chinese string in the widget lua file.

### Steps To Reproduce

1. Add widget to the screen
2. Reboot the radio
3. The widget disappears

### Version

2.10.5

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-12-13:
Does the string have to be anywhere specific? How about a link / sample widget that exhibits this behaviour? ;)

L1cardo, 2024-12-14:
Sorry for my bad memory, I thought I had uploaded the script.

When `drawText` Chinese strings.
[image]

It is just a CN version of ElrsTelemWidget https://github.com/ExpressLRS/ElrsTelemWidget

Here is the file
[ElrsTelemWidget-CN.zip](https://github.com/user-attachments/files/18133064/ElrsTelemWidget-CN.zip)

philmoz, 2024-12-14:
The widget name is too long - maximum of 12 characters:
```Lua
return {
  name = "ELRS Telem CN",
  options = {},
```

L1cardo, 2024-12-16:
Yeah, it is indeed because of the long name, thanks!