EdgeTX/edgetx #5711: Companion: I see the Flightmode panel, although it is disabled

TheIsotopes, 2024-12-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Flight Modes are disabled under Radio Settings -> Enabled Features -> Model Menus but Companion views the Fligh Modes Panel.
also in Editing Model -> Enabled Features -> Model Menus -> Flight Modes stays Global.

@elecpower pls can you check this?

### Expected Behavior

When the Flight Modes Feature is deactivated, i don't want to see the Fligh Modes Panel.

### Steps To Reproduce

see Current Behavior

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

elecpower, 2024-12-15:
Companion and radio do not align in so far as screens and layouts.
If you disable Flight Modes in Companion you would also disable global variables.
So for the time being the radio settings only affect radios.

TheIsotopes, 2024-12-15:
Yes, I was confused by the radio, it is not displayed there.
With the curves, for example, this is only changed on the radio.

ok, closed now.

closed 2024-12-15 by TheIsotopes