EdgeTX/edgetx #5734: Control for the lights of Radios with 6-Pos Buttons

FPVZaphod, 2024-12-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The 6-Pos Buttons of Radios like Jumper T-Pro, T20 or similar can we used for several interesting things. But I cannot control the lights of this Buttons by EdgeTX.
If I as am example press Button 6 to make it active, it's not possible to software wise switch back to the function of Button 1 (and turn on this buttons light).

### Describe the solution you'd like

A function to simulate a button press on this 6-Pos Buttons.
Like "Overide Btn1=sp" for "ShortPress"acting in the same way like I would manually press it. Triggering the same function, and tuning on it's Light.
A use case is as an example to always switch to Button1 when arming. Or to switch to a certain flight mode when RSSI low.
Additionally the buttons lights can be used as status lights, using a new special function "Override Led1=on/off". This would give maximum flexibility.

### Describe alternatives you've considered

I dont see any alternative, as I cannot control the status of the button.

pfeerick, 2024-12-23:
Check out a nightly firmware or a recent main build of Companion... I think
the Push CS special function already does what you want... Programmatically
press a customisable switch button.

FPVZaphod, 2025-01-05:
Cool, thanks - I will take a look! Sounds like the required function!

closed 2025-02-07 by pfeerick