EdgeTX/edgetx #5741: After upgrade from OpenTX to EdgeTX the error "No SDcard" aktiv

Willi-Billi, 2024-12-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a FrSky QX7 transmitter with OpenTX. 
I have made the update to the latest EdgeTX with the Buddy function.
After this, the transmitter start, but have the error "No SDcard" on the display.
The SDcard, a 32Gbyte card is correct on the PC adapter.

### Expected Behavior

I think, when a SDcard works with OpenTX, it will / must also works with EdgeTX.

I was then look for a other, smaler SDCard and this 16GByte it works. But it is not so easy to bay smale SDCard.
Please look, that bigger size will also works.

### Steps To Reproduce

use a 32 GByte SDCard

### Version

2.10.5

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Operating System (OS)

Windows

### OS Version

Win10

Willi-Billi, 2024-12-27:
Dear all,
I have bay a 8 GByte SD card and with this card it works.
Now, I think, is a SW bucg by EdgeTX, that only small card work.
But 32 Gbyte is a small one, there are also 1TB and more in the shop's.
Kind regards
Willi

pfeerick, 2024-12-27:
Some SD cards which do not init properly within the required timeframe are
no longer compatible with EdgeTX as of 2.10. This usually only applies to
generic / clone SD cards, not "name brand" SD cards, and is not related to
size. Larger SD cards will work, as long as you format them correctly as
FAT32 (*not *exFAT). They may also perform better if you create a smaller
(8GB/16GB) partition rather than use the entire SD card.