EdgeTX/edgetx #5752: MASTER/CPPM an MASTER/SBUS not usable

pierrotm777, 2025-01-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I try to use the MASTER/CPPM and MASTER/SBUS as trainer mode.
- I define an input with by example TR6
- I define CH6 with TR6 input
- I connect a PPM signal (positive or negative) on the CPPM
[image]
- I define MASTER/CPPM as trainer mode.

### Expected Behavior

I don't see my channel 6 moving.
I haven't power on the VBAT pin.

I have no issue if i use Master/Jack mode.

### Steps To Reproduce

I have tried to send CPPM Negative or Positive without success.

### Version

2.10.5

### Transmitter

RadioMaster TX16S / TX16SMK2

### OS Version

W11

3djc, 2025-01-05:
From memory I think either HB or SPort pin must work

pierrotm777, 2025-01-05:
In fact, in reality, Vbat is well powered on but I can't use cppm or Sbus input with the S-port pin in modes Master/CPPM or Master/SBUS.

3djc, 2025-01-05:
Did you try heartbeat too ?

rotorman, 2025-01-05:
TX16S JR bay heartbeat pin does not work, see: https://github.com/EdgeTX/edgetx/pull/477#issuecomment-884311379

pierrotm777, 2025-01-05:
Do I try heartbeat for Trainer mode?