EdgeTX/edgetx #5758: enable simulator to use live telemetry data from ELRS

offer-shmuely, 2025-01-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is hard to debug script & widgets that talk to complex elrs communication like MSP 

### Describe the solution you'd like

I like to have ability to run the simulator with live elrs telemetry push/pop

The setup is as follow:
* TX is running with elrs module connected to a flight controller
* computer is connected to TX with usb (usb<-->TX)
* simulator is running lua code in widget or one-time script

requested feature:
* when the code that run in the simulator call crossfireTelemetryPush(), it will be send to elrs module on the TX through the usb
* when the code that run in the simulator call crossfireTelemetryPop(), it will be read from the elrs module on the TX through the usb

wimalopaan, 2025-01-26:
Since we have already a connection to a physical serial port in `simulator` another possiblility would be to allow the simulator use an external connected ELRS-Module directly.

offer-shmuely, 2025-02-07:
Can you explain....