EdgeTX/edgetx #5761: X9D Plus (not 2019) freeze while using/in flight

malarcky, 2025-01-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While using the radio on the last firmware 2.10.5, sometimes it freezes and the screen and buttons wont respond, the screen shows the last thing before the "freeze" but wont update anymore, the sticks and switches still work, but the transmitter its inactive (I still can hear the you abandoned me voice)

### Expected Behavior

Not to freeze randomly

### Steps To Reproduce

I have tried but I cannot reproduce, it happens randomly. I am recording logs. I have tried reflashing, and starting a new sd from scratch. It could be a damaged micro SD?

### Version

2.10.5

### Transmitter

FrSky X9D+

### Operating System (OS)

Windows

### OS Version

Windows 10

pagrey, 2025-01-05:
Does the problem happen when you revert back to the last working version? That would be the first thing I would try. Sounds allot like a hardware issue with the radio to me. There was a discussion about SD card failure on rcgroups and from what I remember a freeze wasn't supposed to happen.

malarcky, 2025-01-05:
Last working version was OpenTX 2.3.2, that was like 5 years ago, I did not use the transmitter on that time, but it was working fine, should I go back to that version?

pagrey, 2025-01-06:
The way I look at it, since I don't do any real coding myself, I do everything I can to help out those that are putting in time to help me with this software. If I wanted to keep using the radio I'd go back to what worked. If it's fine then report back that you are pretty sure the update was probably the issue and be happy you still have a working radio on the old version. If you have problems on the old version then it's probably the radio.

pfeerick, 2025-02-14:
A freeze shouldn't happen, but that doesn't mean it *can't*. If you suspect that could be the case, trying to replicate this, but without logging should help give more data. At least the radio isn't locking up totally, just the UI, from the sounds of things, which means some error/fault handling is clearly working correctly!