EdgeTX/edgetx #5764: "key stuck" message after compiling the latest code to make FW for FrSky Taranis X9DP

pdxtigre, 2025-01-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I got "_key stuck_" message after compiling the latest code from main to make FW for _FrSky Taranis X9DP_.  I verified HW using both "**Analog**" and "**Key**".

I tried to reload "Centurion" 2.10.5, and it was fine.

### Expected Behavior

FW working fine.

### Steps To Reproduce

1. Pull the latest code from main and compile using Docker on Windows 11.
2. FW loaded properly to the Transceiver.
3. Everything frozen after that.
4. Reboot and got "key stuck" message and nothing functioning. 

### Version

Other (Please specify below)

### Transmitter

FrSky X9D+

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Anything else?

Latest code from main showing version 2.11.

pfeerick, 2025-01-12:
Reproducible on X9D+ (and not on X9D+2019) - no hardware keys responding with latest nightly d6f24d1, trims, switches pots all being read.

philmoz, 2025-01-16:
Is it possible the issue fixed by #5787 was responsible?

pfeerick, 2025-01-16:
Quite possible... It looks like the X9D+ would be caught be the default
block where that change was made... will test it again later today.

elecpower, 2025-01-16:
The X9D+ does not have a SYS key unless it is virtual

pfeerick, 2025-01-17:
Yup, looks like that was the culprit... looking for a non-existent key... switch warning and key lockup is now gone for me.

closed 2025-01-17 by pfeerick