EdgeTX/edgetx #5765: Batterie Widgets for LiHV

agmorpheus, 2025-01-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have a small FPV drone that uses a 1s LiHV.
I notice that the battery widget shows a 2s cell because the voltage exceeds 4,2.

### Describe the solution you'd like

It would be nice when the battery widgets get an option where you can set the typ of battery. Depending on the type of battery the cells should be calculated.
LiHV packs become more and more popular so I think it would be a nice improvement.

pfeerick, 2025-01-07:
If you mean the BattAnalog widget, I believe this already has an option for LiHV... try updating your SD card pack ;)

agmorpheus, 2025-01-07:
thanks for the hint. I have the latest version on my card. 
It looks like, that this is only part of the c480x272.zip and not the 480x320.zip (I need this)
Both archives contain completely different lua scripts. 
This is strange.

pfeerick, 2025-01-09:
Thanks for that. I'll close this issue then since this is related to the Lua/SD card widgets, rather than firmware.

closed 2025-01-09 by pfeerick