EdgeTX/edgetx #5772: Compatibility issue - Companion 2.10 - Jumper T15 - External  4 in 1 module.

Gadgetguy2000, 2025-01-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I am using an external 4 in 1 module with a Jumper T15 radio. Radio has the most recent EdgeTX update. Module protocol - When connecting with Companion 2.10 I get an error message "The external module protocol is incompatible" and it has been set to off. Is there a fix for this? (Module V1.3.4.0 AETR - SFHSS protocol)

[image]

### Expected Behavior

I should be able to properly set up my models (including the 4 in 1 module) via Companion

### Steps To Reproduce

1. Connect radio to computer (Windows PC)
2. Set radio in USB mode
3. Launch companion 2.10.5
4. Select one of the models in the radio (I have 3 different models - all need the 4 in 1, all give the same message)

### Version

2.10.5

### Transmitter

Jumper T16, Other (Please specify below)

### Operating System (OS)

Windows

### Anything else?

Please note that this is for the Jumper T15 radio. Thank you!

elecpower, 2025-01-11:
In the T15 Radio Profile what is the External Module set to?

Gadgetguy2000, 2025-01-11:
Thank you Neil for looking into this.

I have three heli profiles.

Two are sfhss, the third one is c186

Neither works

Thank you

Miklos

elecpower, 2025-01-11:
What does each profile display in the area highlighted in screenshot

[image]

pfeerick, 2025-02-09:
@Gadgetguy2000 Can you check the configuration value for the External Module as shown in the screenshot above? The error message you received suggests that it is not set correct for your handset.

closed 2025-03-14 by github-actions[bot]

Gadgetguy2000, 2025-03-14:
I am sorry to hear that this will not work for now , it would have been
amazing. Is it possible a later version will solve the issue?

pfeerick, 2025-03-14:
Please read the earlier replies to this issue, and check the configuration setting for external module - it is most likely you simply need to set it correctly. ;)