EdgeTX/edgetx #5774: Subtrim value moves one limit but not the other

john-c-bryant, 2025-01-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Output is based entirely on one stick.
Min is -50%, max is +50% (want 90 degrees of movement, servo gives 180 degrees for 1000us to 2000us)
Intend to use subtrim to correctly centre range of movement.  Using 10% an an experiment.
All testing being done in Edge Companion.
With "Linear Subtrim" not selected, output goes from -50% to 50% as the stick is moved, and is 10% with stick centered.
This is as expected (subtrim only affect central value).
With "Linear Subtrim" selected, output goes from -40% to 50% as the stick is moved, and is 10% with stick centered.
This is NOT as expected.  Shouldn't the output go from -40% to 60%?
According to the documentation "both the upper and lower limits should shift", and the "reaction of the stick should be the same on both sides of the midpoint".

### Expected Behavior

See above.  I expected the channel output to go from -40% to 60%.

### Steps To Reproduce

Open Companion
Create model with some channel based entirely on one stick.
Set subtrim to 10% , min to -50%, max to 50%, select "linear subtrim"
Simulate and watch channel output as stick is moved from one extreme to the other.

### Version

2.10.5

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

Windows

philmoz, 2025-01-11:
Min & Max are fixed limits - they are not affected by sub trim.

The sub trim is applied first, then the Min and Max limits are checked which is why the upper limit stays at 50%.

You may able to use PPM Center to adjust servo midpoint instead.

philmoz, 2025-01-12:
The limit is not being changed by the sub trim.

The stick value is scaled to fit in the range 'Min' to 'Max'.
Then the sub trim is added.
Then the new value is clipped to remain within the range set by 'Min' and 'Max'.

If the stick is at -100 physically it is scaled to -50 by Min - adding 10 for sub trim becomes -40.
If the stick is at 100 physically it is scaled to 50 by Max - adding 10 for sub trim becomes 60 so it is clipped back to 50.

john-c-bryant, 2025-01-12:
This behaviour is at odds with the user manual, which clearly states that "both the upper and lower limits will shift"?

And, if this behaviour is in fact "correct", just what is the intended point of the "linear subtrim" option?  Note that, if one starts off with symmetrical limits (e.g. 45 degrees either side of centre), one ends up with asymmetrical limits and the stick will be more sensitive on one side of the centre than the other.  So much for "symmetrical" (to quote the documentation).

The choice has to be between fixed limits (the default option) and limits that are relative to the centre point ("linear subtrim").

I can see what you're saying, and agree that this is clearly how the code currently works.  But having one absolute limit and one relative limit doesn't make any sense at all,

philmoz, 2025-01-12:
Yes, the code does not seem to match what is written in the manual; but the code has not changed in 11 years.

The difference is how the physical value is scaled.

In 'Center' mode negative stick values (-100 - 0) are scaled to fit between 'Min' and 'Offset', positive values scale to fit from 'Offset' to 'Max'. So the scaling is different on each side. In your example negative scales from -50 to 10 (a range of 60), while positive is scaled from 10 to 50 (a range of 40).

In 'Symmetrical' mode the stick negative stick values are scaled from 'Min' to 0, and positive is scaled from 0 to 'Max', so the scaling is symmetrical on each side. Then the offset is applied, and then the value is clipped to fit between Min and Max.

Have you tried using PPM Center instead of Subtrim to center the servo?

philmoz, 2025-01-12:
You could also try using 'Center' mode and adjusting your Min and Max by the Subtrim amount.

john-c-bryant, 2025-01-12:
There is no problem in achieving what I need to achieve.  I can simply manually adjust my limits whenever I change the subtrim.  If my ideal starting position is a subtrim of 0% and limits of +50% and -50%, and it turns out that I have to use a subtrim of 10% to centre the servo, then I need on leave "linear subtrim" unchecked and change the limits to -40% and 60%.

The question is whether "linear subtrim" as implemented makes any sense at all.  The fact that code has been the same way for 11 years doesn't mean that there isn't a bug.  It simply means that there isn't a new bug.

If "linear subtrim" were implemented as described in the documentation, it would be handy whenever people want to define their limits in terms of the midpoint (i.e. whenever the upper limit needs to be X above the midpoint and the lower limit Y below the midpoint).  This is my situation (I want the servo to travel 45 degrees on either side of the midpoint), and I'm pretty sure that a lot of others are in the same boat.

As actually implemented, however, the option is useful only if one wants one limit to be absolute while the other is relative to the midpoint.  And furthermore the limit that moves needs to depend up whether the subtrim value is positive or negative.  I can't see this as being useful to very many users.  Indeed I have great difficulty seeing how it can be useful to anybody.  For practical purposes the option might as well not exist,  and I'm reminded of the saying that a "feature" is a bug in a three piece suit.

Code is meant to implement the vision laid down in the manual, and this is especially true when the vision in the manual makes perfect sense and the way the code works doesn't.    I think that by taking the code as it stands as Gospel you're looking at things the wrong way round.

P.S. As its name implies, the "linear subtrim" option doesn't apply when a servo is centered using the PPM option.  So centering using PPM doesn't provide a bug workaround.

philmoz, 2025-01-12:
This code predates my involvement in EdgeTX so I have no idea of the history or why it is the way it is. Since the code was likely written well before the manual it could also just be a misinterpretation when the manual was created.

The risk is that changing the code that has been around for so long could break existing setups in unexpected ways.

john-c-bryant, 2025-01-12:
This is a very valid point, and one that crossed my mind.  If one were to correct what I feel is a bug in the next version of Edge Tx, then any models that make use of "linear subtrim" would have have one of the two limits relaxed, and this could result in servos coming up against physical limits.  Definitely not good.

In an attempt to broaden my understanding, I've had a look at the documentation for Open Tx (see https://doc.open-tx.org/manual-for-opentx-2-2/software-overview/model_menus/outputs).  It describes two options: subtrim affects center only (limits unchanged), and a second possibility that equates perfectly to the way in which "linear subtrim" operates in Edge Tx.  

Basically the whole point is to maintain the slope of the output vs input curve over the entire range of stick movement.   This isn't actually achieved because at some point the output saturates (i.e. the blue line on the graph has two parts, each of which has a different slope), but note that if the upper limit were moved upwards by 20% then the objective could in fact be achieved.  In this case the fact that the upper limit is assumed to be 100% of course precludes moving it upwards, but if the limit was less than 100%, then moving it upwards would be entirely consistent with the avowed objective.  Keeping the limit on the trim side fixed has nothing to do with maintaining a uniform output vs input slope, but instead evidently relates to a philosophical stance on the nature of limits:

"Low and high limits. These are "hard" limits, i.e. they will never be overridden, so as long as they are set so that your servo never forces, it really never will. They also serve as gain or "end point settings", so reducing limit will reduce throw rather than induce clipping."

I think that whether this is a good idea is a debatable point.  I don't know exactly where my limits are - just that they're each 45 degrees from the mid position.  They're just as ambiguous and subject to trimming as my centre position is, and so I'm in no position to enter precise "hard" limits and then leave them in place while I adjust the trim to get the around to get the centre position right.  But in any event it's a clear stance that is clearly laid out in the manual.  And given this stance, one can at least understand the genesis of what I see as very strange behaviour.  It's not that moving both limits isn't the obvious thing to do.   It's just that it's impossible to move both without violating what has been declared to be a fundamental principle.

Evidently you on the right track in saying that whoever produced the Edge Tx manual misinterpreted things.  Basically they dropped the reference to hard limits and, with this out of the way, they were able to present a logically cleaner version of the "symmetrical" option.

The Open Tx manual makes a reference to "center adjustment", while I assume is analogous to the Edge Tx PPM option.

"Center adjustment. This is similar to subtrim, with the difference that an adjustment done here will shift the entire servo throw (including limits), and won't be visible on the channel monitor."

Unfortunately this isn't really explained.  I'm guessing that everything happens exactly as normal until the very last stage before transmission (which would explain why the difference isn't "visible on the channel monitor"), and then at the very last step PPM valves are adjusted by the difference between 1500us and the value selected.  Presumably the green line on the graph isn't quite correct as surely the transmitter has to clip at -100% rather than transmit values below 1000us).

I should try playing around with this.  The fact that nothing is going to show up on the channel monitor makes experimenting a bit harder, but I can do some testing with the actual hardware.

philmoz, 2025-01-13:
The channel monitor is working correctly and shows both the position of the input relative to the defined center point (the graph) and the final transmitted value (above the graph to the right).

Using your example:
[image]

john-c-bryant, 2025-01-13:
I've a transmitter with a monochrome display and the channel monitor doesn't show actual pulse widths.

So I'm looking at two channels showing different percentages, and yet the signal being transmitted is clearly the same in both cases.

I do take your point about output being relative to the "defined centre point", in which case the fact that the channels look different even though the actual output is the same does make some sense.  But how is a newbie like myself supposed to jump to this conclusion?  There's not a word in the manual.  I just made the obvious assumption: that 100% means maximum pulse width (2000 us) and that -100% means minimum pulse width.  And isn't it intuitive that if two channels are outputting exactly the same signal then the channel monitor should show them as doing so?   I do grant that if one is looking at pulse width values this is the case, but it's hard to look at values that aren't there (is there some option I should turn on?).

I've just done a little experiment.  With the limits set to -100% and 100% and the PPM centre set to 1550us the channel monitor shows the actual output going from -100% below the defined mid point to 100% above it.  Unless the monitor isn't reflecting reality, or the transmitter is generating pulses longer than 2000us, this implies that when the PPM centre is moved from 1500us, 100% output is redefined to mean 100% of the distance between the specified PPM value and 2000us (rather than, as usual, 100% of the distance between 1500us and 2000 us).  Which begs the question of what a 50% limit means when the centre PPM is modified to 1550 us.  Does the limit stay at 1750us (i.e. does 50% retain its usual meaning) or does the limit move to up 1775us?

3djc, 2025-01-13:
BW show pulse value in output channel. Select one, you will get µS value for it in top bar.

john-c-bryant, 2025-01-13:
Now I've two questions.

1/. How do I select a channel in the monochrome channels monitor?  Clicking simply moves between the channel and mixer monitors, while rolling just cycles around the four sets of eight channels.  Doubtless it's obvious but I'm afraid not to me.

2/. Exactly what do mean by 50% of travel range?  Is the "travel range" when one sets a limit always the range between 1500us and either 1000us (min limits) or 2000us (max limits).  Or can the "travel range" get redefined to mean the range between the defined centre PPM value and 2000us (for max limits) or 100us (for min limits)?  This isn't quite the "same" thing.

I think that I may be getting my head around things.  In my world view, everything up to the final transmission stages is done in terms to percentages, with 100% representing the maximum pulse width (2000us), 0% representing centre pulse width (1500us), and -100% representing the minimum pulse width.  The view is carried right through to the channel monitor stage (at least as I'm looking at it now), and only at the last moment are the values actually transmitted adjusted to reflect the selected PPM centre value.

I look forward to be able to see PPM values in my channel monitor as I'd like to see what happens when the limits are set to 100% and -100% and the centre value is set to 1550us.  I expect that as the stick is moved down valves PPM values will bottom out at 1050us, and that as the stick approaches its upper limit PPM values will reach 2000us and then stop increasing (i.e. the limits will move with the centre point as advertised, and clipping is inevitable if the transmitter is not to exceed 2000).

3djc, 2025-01-14:
In **output** screen. BW screen does not have the resolution to display it on channel monitor

philmoz, 2025-01-14:
We can fit the PPM value on 128x64 radios:
[image]

john-c-bryant, 2025-01-14:
The output screen does show channel values, but they're not PPM values.  Instead the values simply duplicate the values on the monitor screen (with the additional catch that's it's not easy to tell which channel is which).  So not really very helpful at all.  If it were not possible to include PPM values on the monitor screen, then I think that changing the output screen to show PPM rather than percentage values would be a good alternative.

Can I take it that the next release of Edge Tx will feature PPM values on the channel monitor?

I still can't see the value in displaying the output value  "relative to the defined center point" rather than simply displaying only the actual output value (in % and PPM format, with 1% = (2012 - 1500) / 100 us.   But this is perhaps just me, and I'll be happy as long as I can see the actual output value, which to my mind is all that really matters.

It's interesting to see that the nominal "1000 to 2000us" range is in fact 988 to 2012us.  I suspected that this might be the case, as I'd seen something somewhere, but couldn't find any concrete information when I went looking.  Yet another example of how these days progressing along the learning curve seems to depend upon serendipity, forums, instructional videos, and so on.  Expecting to be able to find everything in the manual is now evidently very old school.

3djc, 2025-01-14:
Then you might not have activated ppm values or run an older version, mine clearly does

john-c-bryant, 2025-01-14:
OK - I'm hooked.  How do I "activate PPM values"?  There's nothing in the manual (of course) and nothing obvious that I can see anywhere in the companion.  

I'm definitely not running an "older version", as I recently installed the very latest release.

philmoz, 2025-01-14:
> The output screen does show channel values, but they're not PPM values. 

Both the outputs list and the output edit screen show the selected channel PPM value.
[image]
[image]

pfeerick, 2025-01-14:
>  How do I "activate PPM values"?  

You can configure how PPM units are shown on current firmware version... SYS/Radio Settings -> Radio Setup -> PPM Unit...  change it to "us" for microseconds view. You'll then see them in channel monitor and main view, etc. 

It should be listed as an the third option for PPM units int the manual here so will need to update that (only references 0 and 0.0 atm).
 https://manual.edgetx.org/bw-radios/radio-settings/radio-setup

john-c-bryant, 2025-01-14:
I'm on the same page now.  As the discussion began with the channel monitor, I assumed that "output page" was a reference to the immediately adjacent page (i.e. screen 4 of the main view), which does also also contain output values.

Now that I realize that I should have been looking at screen 7 of the models view, things are making a lot more sense.  I apologize to 3djc for having wondered whether he was just playing me along.

Yes, the manual does need updating, and it probably won't surprise anybody that I think that it's more than a little strange that an option which ostensibly is there to adjust the display precision of PPM values is also being used to toggle between percentages and PPM values. This is hardly intuitive.   Would it not make a lot more sense to call the option "Output Format" and offer four options: 0us, 0.0us, 0%, and 0.0%?

pfeerick, 2025-01-14:
Probably by this time, but many things in EdgeTX and OpenTX have a long
history. And it would only be three options... There is no 0.0us (and I
don't think there ever would be).

john-c-bryant, 2025-01-15:
The penny has dropped.  Though the option is called PPM Units, and the manual talks about the "level of accuracy for PPM values", it's actually mostly about the precision with which percentage values are displayed.  To quote "Alice in Wonderland" - "curiouser and curiouser".  My guess is that insiders who are have been with Open Tx and Edge Tx for a long time have gotten so familiar with it that things no longer seem strange to them.  But for a complete newbie (albiet one with a lifetime of work in Systems Engineering), some aspects do indeed seem very strange indeed.