EdgeTX/edgetx #5777: Companion - SYS button crashes simulator with some handsets

pfeerick, 2025-01-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

When you press the 'SYS' button with some handsets (only repro'ed with bw handsets so far - RM TX12, RM Pocket, Jumper T-Pro, FrSky X9D+2019. But TX16S was fine) the simulator will crash. 

```
7260ms: pushMenu(1, 0x7ffff3d81fb0)
7320ms: f_opendir(/home/pfeerick/Desktop/test/SCRIPTS/TOOLS) = OK
7320ms: f_closedir(0x7fffcbdffa30)
*** buffer overflow detected ***: terminated

Thread 8 "menus" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcbe006c0 (LWP 14088)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimised out>)
    at ./nptl/pthread_kill.c:44
warning: 44	./nptl/pthread_kill.c: No such file or directory
```
[](url)

### Expected Behavior

It to not do a happy jig and then crash :rofl: 

### Steps To Reproduce

1. Start up a new simulator for an affected B&W handset via Companion
2. Wait for it to start up, press the SYS button or keyboard shortcut

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X9D+2019, Jumper T-Pro, RadioMaster Pocket, RadioMaster TX12 / TX12MK2

### Operating System (OS)

Linux

### OS Version

Mint 22 Cinammon

### Anything else?

Spotted this when testing #5763, haven't tried on Windows yet to see if this is OS or Linux version specific yet.

elecpower, 2025-01-12:
2 steps fwd 1 step bwd

pfeerick, 2025-01-13:
I think you missed a zero there... That would be 20 steps forward in my book :rofl:

elecpower, 2025-01-13:
Tested all radios above and none crash in Ubuntu.

My libsims compiled with debug and run inside my slightly extended version of the official build container. Note: they were not in an app image.

I'll check WinOS MSYS2 builds when I have some time and can fix the toolchain issue you found.

pfeerick, 2025-01-13:
In my case, I was using (and almost exclusively use since it's what the end users use) the AppImage from GitHub.

elecpower, 2025-01-13:
I fear this is going to be a torturous one to nail down and fix :-(

pfeerick, 2025-01-13:
Hm... and Windows build doesn't crash either :( So until otherwise informed, I'm gonna assume it's another of those lovely  "incompatible with newer Linux OS AppImage" style gremlins for now 😆

philmoz, 2025-01-13:
Same error on Ubuntu 20.04.5 with the latest downloaded AppImage.

philmoz, 2025-01-13:
Interestingly it does NOT crash if the 'SD Structure path' in the profile settings points to a folder with actual radio content (sounds, scripts, etc).

It crashes consistently if the 'SD Structure path' is empty or points to an empty folder.

elecpower, 2025-01-14:
Any recent code changes in the SD code?

elecpower, 2025-01-14:
Any recent code changes in the SD code?

The stand alone libsim do not crash if sd path is empty

pfeerick, 2025-01-14:
> Interestingly it does NOT crash if the 'SD Structure path' in the profile settings points to a folder with actual radio content (sounds, scripts, etc).

> It crashes consistently if the 'SD Structure path' is empty or points to an empty folder.

That is sort of what I am seeing here, but also not. i.e. With an existing profile for the TPro that had never had the SD Structure path set, it crashes. If I set the path to a valid path, it no longer crashes. If I clear the SD Structure path (so it is empty again), it no longer crashes... even after a restart of Companion and a restart of the computer. :confused: 

Unrelated bug  ... X9D+ has SYS button (incorrect) as well as Menu button (correct).

elecpower, 2025-01-14:
> Unrelated bug ... X9D and X9D+ sim have SYS button (incorrect) as well as Menu button (correct).

That is a hwdef json build bug

elecpower, 2025-01-14:
Empty crash, set sd and run, clear and run points to corrupt application settings. So we need a crashing config file.

philmoz, 2025-01-14:
This is consistent on Ubuntu 20.04.5 and 24.04.
- Add new profile - radio = Pocket
- Leave SD path blank
- Click '+' button to create new .etx file
- Click 'Simulate radio' (don't add a model)
- Click 'SYS' in simulator - crash
- Restart companion, change the SD path to valid content - no crash when running simulator
- Clear SD path and run simulator again -crash

pfeerick, 2025-01-14:
> Empty crash, set sd and run, clear and run points to corrupt application settings. So we need a crashing config file.

Was about to give you a working and breaking profile, but now I can't reliably reproduce that... the TPro V2 profile I was using before won't fail again, and if I try the same trick with other profiles it will simply crash again once the field is empty... so Phil's sequence and  is the most reliable reproduction now. 

FWIW... Pocket, TX12, TX12 MK2, Zorro, MT12, TPro (SD folder doesn't exist), X9D+, X-Lite S, T-Lite profiles are crashing... but X9D+2019, T-Pro S, T20, and colorlcd TX16S, V16 are not. And if you 'only' have the SCRIPTS folder (and subfolders) present in your SD card "folder" that is not enough to prevent the crash. 

[Companion 2.11 settings.ini.zip](https://github.com/user-attachments/files/18404738/Companion.2.11.settings.ini.zip)

elecpower, 2025-01-14:
Can either of you post your full log so we can peruse the messages to see if there is a clue?

elecpower, 2025-01-14:
Actually would be better to have two logs for same radio for working and crashing to see if that can narrow down the focus.

philmoz, 2025-01-14:
Created a fresh Ubuntu 22.04 VM and installed main branch.
- Release build for Pocket - same crash.
- Debug build for Pocket - no crash (grrrrr!).

pfeerick, 2025-01-14:
Funny you would say that :hand_over_mouth: Same sequence outlined in https://github.com/EdgeTX/edgetx/issues/5777#issuecomment-2588790146 except I didn't create a new profile. 

[pocket-crash.zip](https://github.com/user-attachments/files/18405106/pocket-crash.zip)
[pocket-nocrash.zip](https://github.com/user-attachments/files/18405107/pocket-nocrash.zip)

pfeerick, 2025-01-14:
```
2560ms: pushMenu(1, 0x7ffff3d81ee0)
2620ms: f_opendir(/SCRIPTS/TOOLS) = error 2 (No such file or directory)
*** buffer overflow detected ***: terminated

Thread 8 "menus" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcbe006c0 (LWP 16652)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimised out>)
    at ./nptl/pthread_kill.c:44
warning: 44	./nptl/pthread_kill.c: No such file or directory
(gdb) backtrace
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimised out>)
    at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimised out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimised out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff3e4526e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff3e288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff3e297b6 in __libc_message_impl (
    fmt=fmt@entry=0x7ffff3fce765 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:132
#6  0x00007ffff3f36c19 in __GI___fortify_fail (
    msg=msg@entry=0x7ffff3fce74c "buffer overflow detected") at ./debug/fortify_fail.c:24
#7  0x00007ffff3f365d4 in __GI___chk_fail () at ./debug/chk_fail.c:28
#8  0x00007ffff3f381b9 in __GI___strncpy_chk (s1=<optimised out>, s2=<optimised out>, 
    n=<optimised out>, s1len=<optimised out>) at ./debug/strncpy_chk.c:26
#9  0x00007ffff3d817f3 in addRadioModuleToolHandler(unsigned char, char const*, void (*)(unsigned short), unsigned char) ()
   from /tmp/.mount_EdgeTXpnAifM/usr/lib/companion211/libedgetx-pocket-simulator.so
#10 0x00007ffff3d8229c in menuRadioTools(unsigned short) ()
   from /tmp/.mount_EdgeTXpnAifM/usr/lib/companion211/libedgetx-pocket-simulator.so
#11 0x00007ffff3cf95e7 in handleGui(unsigned short) ()
   from /tmp/.mount_EdgeTXpnAifM/usr/lib/companion211/libedgetx-pocket-simulator.so
#12 0x00007ffff3cf979a in guiMain(unsigned short) ()
   from /tmp/.mount_EdgeTXpnAifM/usr/lib/companion211/libedgetx-pocket-simulator.so
#13 0x00007ffff3cf99f1 in menusTask(void*) ()
   from /tmp/.mount_EdgeTXpnAifM/usr/lib/companion211/libedgetx-pocket-simulator.so
--Type <RET> for more, q to quit, c to continue without paging--c
#14 0x00007ffff3e9ca94 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:447
#15 0x00007ffff3f29c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

pfeerick, 2025-01-14:
Simulator might be having a fit due to changes from https://github.com/EdgeTX/edgetx/pull/5721

I did notice that if there are no tools scripts present, the text stating that only flashes up on the screen for a moment... (on simulator, not sure about handset).

philmoz, 2025-01-14:
I suspect this line:
```C
      strncpy(reusableBuffer.radioTools.script[index].label, label, sizeof(reusableBuffer.radioTools.script[0]));
```

should be:
```C
      strncpy(reusableBuffer.radioTools.script[index].label, label, sizeof(reusableBuffer.radioTools.script[0].label));
```

Testing now.

PR #5780 (2025-01-15): fix(bw): simulator may crash on Linux when SYS key pressed
Fixes #5777
closed 2025-01-15 by pfeerick