EdgeTX/edgetx #5779: Taranis 9XD plus (<2019) delayed power on

villamany, 2025-01-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hello, im trying EdgeTx on my 9XD plus. I'm getting delayed power on after i slide the power switch, it takes about 5 seconds to starting. Its not a SD card issue, without the SD card same issue. 
Maybe the delayed power on function for 9XD plus >2019 is enabled and hidden for this model? 

### Expected Behavior

Not delayed power on

### Steps To Reproduce

Slide the power on button to ON

### Version

2.10.5

### Transmitter

FrSky X9D+

### Operating System (OS)

Windows

pfeerick, 2025-01-14:
> Its not a SD card issue, without the SD card same issue.

This effectively rules out this possibility ... 

> Maybe the delayed power on function for 9XD plus >2019 is enabled and hidden for this model?

... as without the SD card present it can't read the configuration file to then enforce the delay. It should nearly immediately show the "No SD Card" message.  

> I'm getting delayed power on after i slide the power switch

Can you elaborate on this further?  The standard power on delay also brings with it an animation on screen... so if somehow the radio was misconfigured to have the startup delay, you would get a series of four dots on the screen before the EdgeTX logo. If basically "nothing" happens for approx five seconds after turning the power on, I would suggest it is hardware/handset specific issue as I've not seen this on my X9D+'s ... perhaps the RTC battery is nearly flat, and causing some instability during startup? :shrug:

villamany, 2025-01-14:
For some reason my 9XD seems to be not retaining the RTC now, i just replaced the old battery (3.0v) for a new one (3.3v) but still the same RTC reset. However i dont get any RTC battery warning.

But using openTX i dont get this delay issue, also tested edgeTx 2.8.5 and the issue is not present as in v2.10.5.

Video: [https://studio.youtube.com/video/s4rW4BK4WIw](url)

inventor7777, 2025-01-14:
I have the same RTC problem with my MT12!

pfeerick, 2025-02-14:
Still need some clarification for this in order to proceed any further:

> I'm getting delayed power on after i slide the power switch

Can you elaborate on this further? The standard power on delay also brings with it an animation on screen... so if somehow the radio was misconfigured to have the startup delay, you would get a series of four dots on the screen before the EdgeTX logo. If basically "nothing" happens for approx five seconds after turning the power on, I would suggest it is hardware/handset specific issue as I've not seen this on my X9D+'s ... perhaps the RTC battery is nearly flat, and causing some instability during startup? 🤷