EdgeTX/edgetx #5782: RTC time not being kept on RadioMaster MT12

inventor7777, 2025-01-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When the radio is powered off, the time resets to 2000-1-1 even though the RTC battery is 2.97v.

### Expected Behavior

The radio should remember the date and time it was set to.

### Steps To Reproduce

Set the time to anything. 
Power off radio
The time will reset

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

I am (was) running yesterday’s nightly 1/13/24. The problem instantly resolved once I switched to an older nightly which I had on my computer. I saw that someone else had this issue in #5779 with a different radio.

philmoz, 2025-01-14:
If it's a software problem it would help to know which nightly works for you. That might make it easier to pinpoint.

philmoz, 2025-01-14:
FWIW my MT12 does not reset the RTC clock with current main branch.

inventor7777, 2025-01-14:
Oh sorry I guess I didn’t mention the version that works. It’s the 2024-09-30 one.

Gjeremie, 2025-02-10:
Hello
I have exactly the same problem:

- Radiomaster MT12
- the time resets to 2000-1-1 
- RTC batt at 2.95V

Problem happen with the nightly : mt12-36f2b6a.bin (9 february 2025 )

No problem with the nightly : mt12-0ea41f2.bin  ( end of november 2024 i guess)

It is important for me because I have a big LUA script which record log files of my sessions (lap time, telemetry data, ...) named with the date/hour

thanks

Gjeremie, 2025-02-11:
Please can someone could build for us several old nightly firmware for the MT12 (between the Nov 29, 2024 and now )

Dec 6 2024 : https://github.com/EdgeTX/edgetx/tree/3d1e7a914f729fffa77b15d3c3754bfa26f0e6e5
Dec 22 2024 : https://github.com/EdgeTX/edgetx/tree/2b920a014b0d69684eecfc20a74855d8b553e005
Jan 15 2025 : https://github.com/EdgeTX/edgetx/tree/795cc35b8d357825944665f99d824f9c4cd0ca28
Jan 21 2025 : https://github.com/EdgeTX/edgetx/tree/d9f4667cfd410a7124c9765719a55a679a25d96b

In order to we could find when the issue began (testing on my MT12)

We already know that there is no problem before the Nov 29, 2024

thank you

Gjeremie, 2025-02-11:
Ok I isolate the problem :

The issue is in the commit #5693  or #5686  or #5617 or #5699 

Because the commit 0ea41f2 work correctly and in the 3d1e7a9 the RTC time reset to 2000 1 1

@pfeerick can you investigate this please?

thanks

pfeerick, 2025-02-11:
Thanks for whittling that down. 

Won't be #5617 as not related firmware, #5693 and #5699 do not apply to B&W radios. Leaving #5686, which is a likely candidate since it specifically has some RTC changes which will effect the MT21.

3djc, 2025-02-12:
Hmmm, 5686 should be harmless, and is not specifically targeting MT12, but all BW. A DFU flash doesn't involve any Etx code at all, so really intriguing

pfeerick, 2025-02-12:
I suspect the DFU issue is similar to that of the Zorro - i.e. a hardware
issue. With regards to the MT12, ATM I think it is important to just
identify all the possible triggers for the RTC to reset, and then try and
backtrack why it might be happening. e.g. I don't even have to perform a
DFU update... simply plugging in and unplugging the USB is enough to make
it lose the time. And I'm sorry to report, but after removing *just* the
merge commit for #5686, entering the bootloader on my MT12 no longer causes
it to lose the time.

Interestingly, the TX12MKII also lost its time on DFU update (so perhaps
this is an issue across several handset models?, and just needs to be
documented as a known issue since there is nothing we can do about it).
However, going into the bootloader several times didn't result in the time
being lost. And neither T12MAX nor GX12 lost time going into DFU or
bootloader.

3djc, 2025-02-12:
@Gjeremie , could you try #5902 ?

Gjeremie, 2025-02-12:
humm I dont really know the difference
I build the .bin file then flash using this method:
https://manual.edgetx.org/installing-and-updating-edgetx/update-from-an-earlier-version-of-edgetx-using-the-bootloader

Gjeremie, 2025-02-12:
> You should try with DFU (connect USB while radio is off). You can use Companion to do it using Read/Write then Write firmware to radio

OK i will try this tonight

Gjeremie, 2025-02-12:
@3djc , another question: I only use these options to build the firmware :

 -DPCB=X7 -DPCBREV=MT12 -DDEFAULT_MODE=2 -DCMAKE_BUILD_TYPE=Release

Is this OK ?

Gjeremie, 2025-02-12:
ok @3djc , I flashed via DFU your #5902 and:

at first power on (after the flashing) the time is ok

at second power off/on cycle the time is reset 

So doesnt work sorry

3djc, 2025-02-12:
Could you try with a new battery ? (Yes I know it works with older versions)

pfeerick, 2025-02-12:
> @pfeerick ,how did you manage to make it work ? Because I tried with a custom build based on the current 2.11 RC1 and just removing the corresponding line of the https://github.com/EdgeTX/edgetx/pull/5686 in the code (main.cpp and Cmakelists.txt) and it doesnt work (issue is always there)

I just reverted the merge commit for that PR:

`git revert 0ed1b94`

btw,

`FLAVOR=mt12 tools/build-gh.sh` is much easier and less error prone if you want to build firmware (if it is a one-off build, rather than when making incremental changes).

Gjeremie, 2025-02-13:
### **I made other test, to summarize :** 

**- flash with the  mt12-0ea41f2.bin (2.11 november 2024 nightly):**
**.bin file already built downloaded on the website**

          At each power off/on of the radio : No problem at all

**- flash with the  mt12-f51c67a.bin (2.11 RC1):**
**.bin file already built downloaded on the website**

          At each power off/on of the radio : the date/time reset to 2020 1 1   0:0:0

**- flash with the  mt12-f51c67a.bin but reverting the commit #5686  (2.11 RC1 without #5686 ):**
**.bin built with this tutorial : https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod**

          At each power off/on of the radio : the date/time reset to 2020 1 1   0:0:0

**- flash with the  mt12-f51c67a.bin but reverting the commit #5686  (2.11 RC1 without #5686 ):**
**.bin built with this command line: FLAVOR=mt12 tools/build-gh.sh **

          At each power off/on of the radio : the date/time is conserved
                                                                   BUT, during my test, at about the seventh power on/off cycle the date/time was reseted

inventor7777, 2025-02-13:
On a somewhat unrelated note, I decided to build the Dec 6 nightly (https://github.com/EdgeTX/edgetx/tree/3d1e7a914f729fffa77b15d3c3754bfa26f0e6e5) for myself to get the Set Screen function as I thought the issue might not be present, but nope, after the first restart it loses the time completely 🫠

Back to my trusty 9/30/24 nightly :)

Gjeremie, 2025-02-13:
@inventor7777 , try this one:  https://github.com/Gjeremie/edgetx-JG-2.11-RC1/tree/revert-5686-3djc/bw-f4-em/Firmware
it's the 2.11 RC1 but without the #5686 

and it's seems to work ok with no time/date reset

inventor7777, 2025-02-14:
@Gjeremie 
I tested it and it seems to work great! I restarted the radio a bunch of times with no reset 🥳

However, oddly enough, the radio says it’s running EdgeTX 3.0.0….

Gjeremie, 2025-02-14:
What is the RTC backup used for?

3djc, 2025-02-14:
When things go bad (could be software issue, interference, hardware glitch), a hardware watchdogs resets the radio and signals something bad happened. When restarting, EdgeTX detect the "something went bad" info, and expedite an emergency fast start (things like switch position and all are skipped) which goal is to give you bad control of your model. You can identify this condition, if it ever happens, by a blinking '!' character top left of MT12 screen. RTC Backup plays a key role in that

pfeerick, 2025-02-15:
> However, oddly enough, the radio says it’s running EdgeTX 3.0.0….

Which means that is from the main branch, not 2.11 branch ;) Main and 2.11 are pretty much identical atm though. 

> Removing RTC Backup is not world best idea,

True, but given it wasn't actually enabled for B&W F4 until now, it's also not that much of a loss 🤭 However, in my view omitting #5686 is only a workaround, while we figure out a solution (assuming it is a soft bug).

3djc, 2025-02-15:
Still, between the risk of loosing time, and the risk of loosing models, I would choose to fix the later, and to some extent is a tribute to show how stable BW radio are software wise, EM are quite rare on those. And yes, it's clearly not a hard bug since it is only affecting few users

Gjeremie, 2025-02-15:
I hope I never experienced an EM on my MT12 but how long does it take to recover the control of the model?

hawku, 2025-02-15:
> [@inventor7777](https://github.com/inventor7777) , try this one: https://github.com/Gjeremie/edgetx-JG-2.11-RC1/tree/revert-5686-3djc/bw-f4-em/Firmware it's the 2.11 RC1 but without the [#5686](https://github.com/EdgeTX/edgetx/pull/5686)
> 
> and it's seems to work ok with no time/date reset

Just tested 2.11-RC1 on my MT12 and got an RTC reset after the first power cycle.
@Gjeremie 's firmware doesn't reset the RTC after power cycle.
Original "MT12-edgetx,2.10.0-12232023.bin" firmware doesn't reset the time
Radio indicates the RTC voltage as 2.69V, so my results might not be that reliable.

inventor7777, 2025-02-15:
So I tested the latest 2/14/25 Nightly with *both* of my MT12s (one is a very, very early model that I had imported back in late 2023 when they were first released, and the other is a standard one I bought a few months ago) and they both reset the time after a minute of being off. I then built the exact same firmware in Gitpod with `git revert 0ed1b94 ` and the problem instantly disappeared.

pfeerick, 2025-02-15:
> I hope I will never experience an EM on my MT12 but how long does it take to recover the control of the model?

Hard to quantify, as it is may also be a question of how long does it take for the RF link to re-establish. Firmware itself will usually recover in well under a second on B&W... faster if it can use the RTC backup memory and skip SD card re-init. 

> And anyway EM is not acceptable for a radio. It **should not** occur during a race. Never have one on my old sanwa or futaba radio.

Any why does the radio care about whether you are racing or not? 🤪 EMs are very rare, and outside of the odd software bug being the culprit, IMO tend to be somewhat hardware related (brownout, storage failure, etc) or a Lua script somehow doing something out of bounds or some other user-induced gremlin. But the realistic alternative is a complete handset lockup, or a handset with no Lua, no customisation of sounds, limited configuration options overall, etc, etc.. neither of which are desirable.

inventor7777, 2025-02-15:
I can vouch for the reliability of these radios. I’ve been using my MT12 on average 4 times a day since late 2023 and I’ve customized _everything_ and I’ve never had any serious issues, except one that’s kind of my fault. I even got rainwater inside mine and it worked fine (until I tried to turn it back on later, lol) (I fixed it btw)

pfeerick, 2025-02-16:
> I even got rainwater inside mine

I'm sorry, we don't have exception handling for that 🤣 My TX16S and other handsets have had some light rain exposure over the years and have thankfully coped after being left to dry out..

adamalbarr, 2025-03-06:
> So I tested the latest 2/14/25 Nightly with *both* of my MT12s (one is a very, very early model that I had imported back in late 2023 when they were first released, and the other is a standard one I bought a few months ago) and they both reset the time after a minute of being off. I then built the exact same firmware in Gitpod with `git revert 0ed1b94` and the problem instantly disappeared. 
> 
> 
> You can download my custom firmware here - https://github.com/inventor7777/EdgeTX-RTC-Fix/releases/tag/3.0

This will replace the firmware on the device ? 

So i have to download and flash via pc or can be flashed on the air?

inventor7777, 2025-03-06:
@adamalbarr 
Yes, it’s a complete firmware upgrade that you can install via PC. Personally, I use EdgeTX Buddy’s Local File function as its very fast and reliable.

adamalbarr, 2025-03-07:
@inventor7777 
Okay!! i will try it soon 💨
Need rtc baterai replacement first, is bcs my rtc bat is only 1.7v left.
Jst eliminate one by one the causes of the problem.

inventor7777, 2025-03-07:
@adamalbarr 
Wow! Down to 1.7v already? I’m surprised! I’ve had mine now for nearly 1.5 years I think and it’s still at 2.7 iirc.

inventor7777, 2025-03-14:
Wow! You’d have to ask the main EdgeTX devs if it would be possible to add CMOS…idk as my standard battery has lasted a long time. It’s still 2.69v after some years

3djc, 2025-03-27:
Could someone test this ?

[MT12-RTC-TEST.zip](https://github.com/user-attachments/files/19484440/MT12-RTC-TEST.zip)

inventor7777, 2025-03-28:
I left a comment on PR

arkarkark, 2025-03-31:
I had this problem and indeed my RTC battery was flat. I was intimidated about changing it but it was really easy. Just remove the two 2.5mm hex bolts by the steering wheel, slide that mechanism out whilst being careful of the ribbon cable and you have easy access to the battery. just flex the little metal strip holding it in and it'll pop out. Its size is CR1220.

inventor7777, 2025-03-31:
I’ll try buying and changing out the batt soon. Mine still has 2.5v so I figured it was enough…?

3djc, 2025-03-31:
> I’ll try buying and changing out the batt soon. Mine still has 2.5v so I figured it was enough…?

2.5v is already very depleted

pfeerick, 2025-03-31:
It certainly is if that is the resting voltage (as it will drop as soon as it is loaded). Do we know how far off the general ratings of 1.65V - 3.6V is real-world usable for this handset?

[image]

@inventor7777 was the 2.5v voltage the one shown on the handset when at the radio -> hardware page (i.e. when it has a burden load placed on it for measurement)? Or did you measure it some other way?

inventor7777, 2025-03-31:
@pfeerick 
My battery is 2.56v when measured from Hardware page. 

[image]

FWIW I had the same issue on my dad’s MT12 initially (2.99v) but I haven’t tested @3djc’s custom firmware on it as he doesn’t like to have issues like that 😆  (I reverted him back to my custom built firmware with the removed RTC backup)

I will order the CR1220s today and will update back once I install it.

inventor7777, 2025-04-03:
@3djc @pfeerick 

I replaced the CR1220 (it literally only took me about 2 minutes 🤯) and now everything seems to work fine.
I used a name brand heavy duty battery and the voltage reading on the radio is 3.22v.

Do you think that the RTC warning threshold (at least for the MT12) might be a _little_ high? I would have expected the radio to warn me before the battery went flat. (Check RTC was checked as you can see in the photo above)

chrislavender, 2025-04-03:
Been watching this as I'm having the same issue. Surprised I can't find any documentation regarding where/how to replace the RTC battery? Nonetheless, I also have `Check RTC` enabled and hadn't received any warnings. Mine is currently reading at 2.72v.

inventor7777, 2025-04-03:
@chrislavender 
RadioMaster should probably add a section in their manual about it.

All you have to do is remove the two 2.5mm screws from the steering wheel assembly, pull it away carefully (there is a ribbon cable). On the main board inside the radio, there’s a brown enclosure which has a CR1220 inside it. Use the hex tool to push aside the metal tab on top of the battery, which will then pop out with high velocity 😄

Then you can just click the new battery in place, then reassemble. (Be careful about the ribbon cable!)

inventor7777, 2025-04-04:
@pfeerick @3djc 

Welp, the problem is still there. For whatever reason, it always seems to happen after I’ve been doing a lot of driving, then the next time it boots there is no time. I guess I’ll just build firmware with no RTC backup at all from now on…

inventor7777, 2025-04-04:
It’s so bizarre how it doesn’t reset if it’s just sitting doing nothing, then it eventually resets anyway.

inventor7777, 2025-04-04:
My dad’s MT12 which is running my custom firmware w/ no RTC backup at all is working flawlessly. I’ve never messed with the time for 2 months now.

inventor7777, 2025-04-16:
Update - Yesterday I built and installed the latest nightly without any RTC backup, and the issues instantly disappeared. 

Oddly enough, the time reset after the first restart after the update, but I re-set it and no issues since.

pfeerick, 2025-04-17:
> Oddly enough, the time reset after the first restart after the update, but I re-set it and no issues since.

If you did a DFU update, that seems to be normal 🤷

inventor7777, 2025-04-17:
That makes sense. I figured that was the case after I wrote the comment 🙃

druckgott, 2025-05-03:
I also have this problem on the Jumper T15 several times with smaller <2.11 (no experince with 2.11 because I did the update 5 mins ago ;-))

c-lega, 2025-05-05:
Same here with MT12, after changing battery (reported voltage > 3.3v).
Also did firmware re-flash (2.11.0) and sd card (also tested new one - no changes).

3djc, 2025-05-05:
@pfeerick , given the reduced risk linked to the type of models controlled by mt12, maybe we should disable rtc backup for this one

pfeerick, 2025-05-07:
@3djc Yeah, I think that is probably the best way to resolve this for now since it seems the MT12 is only one to really have this issue badly (the zorro RTC issue was different, and hardware related). Given RTC backup was never enabled for B&W STM32F4  prior to 2.11, and that there haven't be a host of reports of issues around EMs, I'm not overly concerned about leaving it off (although I might question "reduced risk" if I get hit by a 1.8kg RC car going at 45km/h+ :-P)

LTabora, 2025-05-07:
I have the Zorro and now the MT12 issue 😫.

I understand a capacitor soldered to the Zorro solves that one. Will wait for the update to the MT12.

Thanks for all your help guys!

druckgott, 2025-05-07:
Jumper t15 also

inventor7777, 2025-05-07:
I think that’s the best option too. I’ve never ever ever had EM, even when driving my 5.5kg Limitless at 186km/h ;)

LTabora, 2025-05-22:
> update

@pfeerick , after the latest update  (2.11.1), both my Radiomaster MT12 and Zorro are now not having any issues with the RTC. 

I thought the Zorro was a hardware issue, and I was about to solder the capacitor proposed by another user, but I've charged it, used it, synced it with companion, and tinkered with it and the RTC continues without a glitch. 

It would seem like some Zorro's were being affected by the same issue as the MT12 🤷🏻‍♂️?

pfeerick, 2025-05-23:
It is entirely possible, as the Zorro has a STM32F4 MCU, however #6223 **only** applies to the MT12, and does not alter anything for the Zorro. Also, if the issue was present in 2.10 (for the Zorro), there is something entirely different happening.

[3 more replies]

closed 2025-07-16 by inventor7777