EdgeTX/edgetx #5784: [Surface/MT-12] Throttle trim has no effect on the reverse throttle range

Divefire, 2025-01-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I saw this mentioned by @rotorman on the Discord regarding the behaviour of the throttle trim "There must have been some reasoning behind limiting trim on MT12 to forward portion only, as the manual says:
`On surface radios (EX: MT-12), the throttle trim has no effect on the reverse throttle range and has some effect over the entire forward throttle range.`
If you are experienced surface model user, please open a GitHub issue and provide your point how this should be handled, or if you are able to code C++, just hand in a pull-request with the changes how you would like to have it."

### Describe the solution you'd like

So thought I'd start the discussion on it. I'm struggling to find a reason why you'd want that trim limited in such a way. I'd actually think on nitro powered surface vehicles you would want the extra fine tuning of the servo so our brake or throttle isn't coming in too soon. Though I suppose you can adjust that via mechanical linkage. 

On the electric side the ESC would limit the need for triming the centre point, but you could potentially give yourself more throttle throw by adjusting the trim point backwards. 

So I'm for implementing the trim back on to the break/rev side of the throttle.

pfeerick, 2025-01-15:
Er... huh? I can trim both fowards and reverse on MT12 with ETX 2.10.4/2.10.5 ... basically 7% forwards and 7% backwards. I wonder if there has been some confusion over the throttle idle setting... Do you have the MDL -> Setup -> Throttle -> T-Trim-Idle option enabled by any chance? As that will configure the throttle trim to only control the "idle" range - on a aircraft this would be when the throttle stick is fully down, and the equivalent for surface was considered to be neutral / centre throttle... Per https://github.com/EdgeTX/edgetx/issues/4504

Divefire, 2025-01-15:
No, you're quite right. I got myself in a muddle. I'll delete this as it's surplus, sorry to waste your time on it.

closed 2025-01-15 by Divefire

pfeerick, 2025-01-15:
Phew... I was worried there for a minute! :laughing: All good, glad you figured it out...