EdgeTX/edgetx #5786: Jumper T15 Radio Info top bar widget dont work

malarcky, 2025-01-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On Jumper T15, top bar "Radio info" widget only shows full battery and full radio strenght, only volume or logging status work. Battery is calibrated.

### Expected Behavior

It should work as is no other setting in the widget.

### Steps To Reproduce

Add the widget to the top bar.

### Version

2.10.5

### Transmitter

Other (Please specify below)

### Operating System (OS)

Windows

pfeerick, 2025-01-15:
I'm not sure exactly what you mean here... 

"On Jumper T15, top bar "Radio info" widget only shows full battery and full radio strenght, only volume or logging status work. Battery is calibrated."

The radio info widget does only show battery, signal strength (when using a telemetry capable RF link), volume level, logging status (and GPS, if supported). If you meant the click is missing, that is a separate widget - Date Time.  

Did you mean to say that only part of that is working? If so. what part? If you mean it is always showing fully battery, check what you have the battery meter min and max values set to in Radio Settings -> Hardware. They should be something like 6.2 - 8.2. As far as the signal strength, that depends on the link.

malarcky, 2025-01-15:
It always shows full battery and full radio strenght, as I say battery is correctly calibrated, it still show its full even when empty, the same happens with the radio.

pfeerick, 2025-01-15:
I can't tell why the RF signal bar is not working correctly without knowing more about your setup  - i.e. what protocol, what tests have you done to demonstrate it isn't working... turn the receiver on, full signal, turn the receiver off, still full signal (which would clearly be impossible!)... 

For the battery - I am not talking about the calibration - but the *range* of the meter... if you have that set incorrectly, of course the gauge is always going to be full. With respect to the screenshot below, (1) is the low value for the gauge and (2) is the high value - for which it will then determine how many battery segments to display. And (3) is the current "calibrated" value. If you have the range configured correctly, and the battery is say around 7.2 volts, you should have three of five segments showing on the gauge. 

Is the voltage correctly updating in Radio Settings -> Hardware? The calibrated value shown (which doesn't refresh on it's own - you need to switch from one tab to another to get it to update, or scroll the rotary encoder while editing it) shown is the current voltage of the battery as read by the firmware. So if that is not correct for some reason, we need to determine if it is a software issue or a hardware issue. 

[image]

malarcky, 2025-01-15:
Okay... problem solved, the theme I was using has the same colour on the two states on that widget, its Dark Theme from the official pack.

closed 2025-01-15 by 3djc