EdgeTX/edgetx #5793: Mixer without source, kills following mixes (no more output)

mama99de, 2025-01-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If for example an mixer **without Source** is set up, all following mixes are not working any more
[image]
In this case, starting from mix 6 no more output

**Mixers without Source, can only be Set up using Companion**, on the TX itself it is not possible to setup a mix without source

### Expected Behavior

Mixer without source should not kill all following mixes

### Steps To Reproduce

Just set up any mixer without source in Companion like here:
[image]
and mix starting from CH6 will not work any longer

Can be produced using the FRSKY Xlite or on my RM Pocket, as well as in simulation in Companion, 

### Version

2.10.5

### Transmitter

FrSky X-Lite / S / Pro

### Operating System (OS)

Windows

### OS Version

Win 10

### Anything else?

**Mixers without Source, can only be Set up using Companion**

elecpower, 2025-01-16:
No source is permitted in Companion to allow conversion between incompatible radios without dropping the whole mixer line. The mixer line should either be deleted or a valid source added for compatibility with radio firmware.

elecpower, 2025-01-16:
Suggested change to Companion is for mixer lines without source to be dropped on writing to yaml files since the user has had an opportunity to fix after conversion but failed to do so.

If they want the dropped mixer lines back they only need to reconvert and add a valid source before writing.

pfeerick, 2025-01-16:
Sounds good, as it's *NOT* valid for the firmware.

philmoz, 2025-01-16:
> Suggested change to Companion is for mixer lines without source to be dropped on writing to yaml files since the user has had an opportunity to fix after conversion but failed to do so.

Maybe add a warning that there are invalid mixer lines that will be deleted and let the user cancel the operation.

elecpower, 2025-01-17:
I suspect inputs suffer from the same issue

PR #5798 (2025-02-06): fix(cpn): inputs and mixes must have source
Fixes #5793

Summary of changes:
- check every model's inputs and mixes for source
- new traffic light status for each models and settings file, green if no errors
- if no source(s):
  - only allow saving to etx format
  - disable writing to radio or sd path as no source causes unexpected behaviours
  - disable simulate model and radio for the same reasons
  - new dialog to display errors
  - change status to red and display the number of models with errors
closed 2025-02-06 by pfeerick