EdgeTX/edgetx #5805: GX12 model.yml from 2.10.2 T-pro v2 does not work on GX12 running

arkarkark, 2025-01-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

when I copy a model.yml file from my v2.10.2 jumper t-pro v2 to a radiomaster GX12 it loses all the inputs mixes and special functions and logical switches

also when I just copy one working model.yml file to another on my GX12 mounted as a USB drive to my m1pro mac it sometimes also loses a lot of the settings and sometimes the `file` command describes the file as 
`MODELS/model07.yml: ASCII text, with CRLF, CR line terminators`
rather than the usual `MODELS/model06.yml: ASCII text, with CRLF line terminators`

[fromtpro.orig.txt](https://github.com/user-attachments/files/18467683/fromtpro.orig.txt)

### Expected Behavior

I should be able to copy a yml file from a t-pro v2 to my gx12,it should load it, upgrade it to 2.11 and write it to the disk and not lose any settings.

I should also be able to copy one model.yml file to another new model.yml file while the gx12 is mounted as a usb drive and it shouild just copy the file.

### Steps To Reproduce

1. copy the attached file to a GX12 and see if it has inputs and mixes

also
 
1. on a gx12 mounted as a usb drive with a command line in the sdcard root directory
5. `cp MODELS/model00.yml MODELS/model33.yml`
6. `file MODELS/model00.yml MODELS/model33.yml`
7. they should both be `ASCII text, with CRLF line terminators` and not `ASCII text, with CRLF, CR line terminators`
8. try copying the file a few times, it doesn't happen every time. and also try 
9. `cat MODELS/model00.yml >MODELS/model34.yml` sometimes that *does* work

### Version

2.10.5

### Transmitter

RadioMaster GX12

### Operating System (OS)

macOS

### OS Version

Sequoia 15.2

### Anything else?

I'm sorry this is a little vague, I've been copying models around and it all works in the UI, buy copying files around on the SD card oftentimes leads to corrupted model files. hopefully you can easily reproduce with these steps, especially my model.yml file attached.

arkarkark, 2025-01-19:
forgot to add the GX12 is running this firmware from the factory
2.11.0-selfbuild . 
GIT# d5552c80 
2024-10-04 10:17:34

arkarkark, 2025-01-28:
this might be related to #5841

MEBeans, 2025-01-30:
Arkarkark.... How do you get the sd card folders inserted into this thread? I get a ".yml type not allowed". I am on windows platform and I am getting the same inputs and mixes as well as channel order cleared out whenever usb is involved.

3djc, 2025-01-30:
Your sd content doesn't matter. Is this also happening when you use ONLY usb joystick?

MEBeans, 2025-01-30:
I'm in the process of testing that now....

MEBeans, 2025-01-30:
I just connected using joystick and flew in Liftoff 3 times. All configs stayed put after disconnecting from the PC. The reason I was asking about getting folders into this thread is because I was asked to provide copies of my /RADIO and /MODELS folders to get any assistance with this.

pfeerick, 2025-01-30:
> How do you get the sd card folders inserted into this thread? I get a ".yml type not allowed".

You can either change the file extension to something that is accepted, like .txt, or better still, since you want to upload multiple files, create a zip file with the contents of RADIO and MODEL folders.

arkarkark, 2025-01-30:
> Arkarkark.... How do you get the sd card folders inserted into this thread? I get a ".yml type not allowed". I am on windows platform and I am getting the same inputs and mixes as well as channel order cleared out whenever usb is involved.

I rename the file .yml.txt then I can attach it

MEBeans, 2025-01-30:
Thank you for that info, I really need it right now! I was just flying my Air75 around my room for a couple hours. While flying, I was figuring out the telemetry. I made a couple adjustments to the telemetry displays. My flight batteries needed charging, so I powered down for a bit. I came back 10 minutes later and powered the radio back on..... all the settings for the model I was just flying are cleared out, this time without the usb being involved! This is getting REALLY frustrating!! I am going to see about getting my radio configs submitted here. Maybe someone will see something?

hawku, 2025-01-31:
> Thank you for that info, I really need it right now! I was just flying my Air75 around my room for a couple hours. While flying, I was figuring out the telemetry. I made a couple adjustments to the [...]

I had the same problem with GX12. Models got corrupted randomly after restarting the radio. That happened multiple times a day. Tried to remove .luac files and every other trick I could find from EdgeTX discord or Github.

Finally I got the latest nightly version of EdgeTX Companion Windows app and used that to read the settings from radio, edited few values and wrote them back to radio. I also removed all models from radio's backup directory.

Haven't had any corruption in 6 days and I am still using the same models.
I don't know why those steps fixed the problem, but maybe there is something wrong with the YML file reader/writer?
All corrupted YML files got cut about the same position, maybe at 1024 bytes.

hawku, 2025-01-31:
Now everything is broken again...
I tried to install Betaflight LUA scripts and radio got corrupted again.
Radio.yml and model00.yml reseted to default settings after copying LUA scripts to radio, safely removing the USB device from windows, removed the USB cable from radio and cycled the power.

3djc, 2025-02-06:
I assume this is a duplicate of #5841

closed 2025-02-06 by 3djc