EdgeTX/edgetx #5810: GX12 hangs while flying

arkarkark, 2025-01-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While flying my model called Pumpkin (model02.yml in attached zip file). My Radiomaster GX12 has got into a locked state where luckily the sticks still control the plane, but none of the other buttons/screens work. everything appears locked up. 

This has happened three times over two days flying. The only way to get my radio back appears to be to pull the battery out and put it back in.
I moved this model over from my Jumper T-Pro v2 (I have another issue #5805 about that not being the smoothest process)

[gx12.2025-01-20.zip](https://github.com/user-attachments/files/18484371/gx12.2025-01-20.zip)

here is a [video of the radio when locked up](https://youtu.be/hSKykFWF1Sc)

here's the log file from a time that it locked up
[Pumpkin-2025-01-20-103633.csv](https://github.com/user-attachments/files/18484437/Pumpkin-2025-01-20-103633.csv)
I've also made this cvs available in a [google sheet](https://docs.google.com/spreadsheets/d/1oBGobEJg8P7Mp5u76gwll8M6o_46H_yvexkBabhjdFE)

luckily this was all line of sight, but if it had happened when I need the switches to change flight modes or return to home I would be stuck.

1. is there a way to reboot the GX12 radio (when holding the power button doesn't do it) without pulling the battery out?
2. it's possible that some missing telemetry scripts are casuing the problem? I had forgot to copy them over from my old radio. I. have put them on now and if I don't experience the problem next week I'll comment back here. The scripts are TOOLS/GPS Stats X9L.lua and Telemetry/GPSx9L.lua from [moschotto/OpenTX_GPS_Telemetry}(https://github.com/moschotto/OpenTX_GPS_Telemetry/tree/main/x7_x9lite)

The plane is a scythe delta wing with a cheap ELRS 2.4Ghz rx connected to a matek 6.ch  CRSF to pwm board and a cheap GPS unit attached to the second uart.

the GX12 is running this firmware from the factory
2.11.0-selfbuild .
GIT# d5552c80
2024-10-04 10:17:34

### Expected Behavior

radio should not lock up

### Steps To Reproduce

1. fly plane with a GX12 transmitter. 
2. at some point notice that radio isn't talking to me. work out no buttons or switches work on the radio. on the sticks.

### Version

Other (Please specify below)

### Transmitter

RadioMaster GX12

### Operating System (OS)

macOS

### OS Version

sequoia 15.2

pfeerick, 2025-01-21:
Since you are running factory/pre-release firmware, can you try updating to a nightly firmware for this radio, and see if this is any different. Double check first if there is a copy of the firmware saved in the `FIRMWARE` folder on the radios onboard memory, as then you can revert back to that if you have any issues, and there isn't one, I'll ask RM for a copy of what they are shipping with.

pfeerick, 2025-01-21:
> is there a way to reboot the GX12 radio (when holding the power button doesn't do it) without pulling the battery out?

If it is *just* a UI lock up, holding down the power button for about 10 seconds should forcibly power down.

arkarkark, 2025-01-21:
> > is there a way to reboot the GX12 radio (when holding the power button doesn't do it) without pulling the battery out?
> 
> If it is _just_ a UI lock up, holding down the power button for about 10 seconds should forcibly power down.

ok. I tried holding it down for a lot longer than 10 seconds and it didn't power down. I show it for a while in my video, but also tried it for much longer off camera. I'll try updating to a nightly build.

pfeerick, 2025-01-21:
Yeah, that looks like 15 seconds easily... ouch! Hm... strange... if you have controls working, mixer is working, so the `isForcePowerOffRequested()` check should have caught that...

arkarkark, 2025-01-21:
Thanks, I have the nightly on there now so I'll be able to report back if it still has this problem with the nightly. but with the nightly my little lua script doesn't run. I asked in the [discord](https://discord.com/channels/839849772864503828/842693696629899274/1331141581000019988) I'll file am issue if needed. It sure is nice to have companion to set up this radio (and my MT12 too!)

pfeerick, 2025-01-21:
You may just need to delete the `.luac` file to get your script working
again... so that it is recompiled for the new lua interpreter.

arkarkark, 2025-01-25:
I tried it again today and it locked up on my talon model, luckily before I had taken off. I was running the c4646cf0 nightly gx12 firmware. (md5sum 6dbe68f08b86987c2c81183809e2b7e5)

philmoz, 2025-01-26:
I've flown at least a dozen times with the GX12 on various builds of 2.11 without any issues.

In you .zip file model02.yml is loading a telemetry screen using a script name GPSx9L which isn't in the zip file as far as I can see. Can you upload this script if you are still using it.

arkarkark, 2025-01-26:
I mention GPSx9L in my original report. it was indeed not present.
I had copied over the .yml files from my old t-pro v2 and neglected to copy the telemetry scripts. but I had since added it and today the lockup happened with that file present.
Usually the problem happened with my plane called "Pumpkin", but today it happened with "Talon"
The GPS telemetry was all working really well today, so that was nice.

arkarkark, 2025-01-26:
@philmoz I wonder if you have telemetry logging turned on with "SD Logs" special function?
It's only a feeling, but I suspect writing to the flash storage is not as reliable as it could be.

I've also flown my GX12 more than a dozen times with no issues, but I've also flown 4 times where I've had to pull the battery out before I could get my radio to work again. I even have video of it locked up.

philmoz, 2025-01-26:
Things to try:
- check radio storage for errors (mount via USB and run a disk check)
- turn off logging and remove the GPSx9L telemetry screen (which also does logging).
- run bench tests to see if the problem is still reproducible

arkarkark, 2025-01-27:
I ran fsk.fat on my linux system and it found filesystem errors. I then fixed it. and checked it again on my mac and all was well.

```
sudo fsck.fat -n  /dev/sdc1
fsck.fat 4.2 (2021-01-31)
/MODELS/._model00.yml
  Duplicate directory entry.
  First    Size 4096 bytes, date 21:07:02 Jan 20 2025
  Second   Size 4096 bytes, date 21:07:06 Jan 20 2025
  Auto-renaming second.
  Renamed to FSCK0000.000
/SCRIPTS/TOOLS/GPS Stats X9L.lua
  Duplicate directory entry.
  First    Size 6438 bytes, date 18:52:36 Jan 20 2025
  Second   Size 4680 bytes, date 18:52:38 Jan 20 2025
  Auto-renaming second.
  Renamed to FSCK0000.000
Label in boot sector is 'WTWFDOTCOM', but there is no volume label in root directory.
  Auto-removing label from boot sector.
Reclaimed 3 unused clusters (98304 bytes).

Leaving filesystem unchanged.
/dev/sdc1: 1427 files, 2165/15365 clusters
```

But then I copied some model files back and over various mounting an unmounting my bee model got corrupted. Then I copied the bee file back and my diana jet model was corrupted. and I copied that back and my bee model is corrupted again. I am just using terminal `cp` on my macbook pro.

Next step is to format the sdcard and try again I guess.

philmoz, 2025-01-27:
Does the same thing happen if you start in bootloader mode and connect the USB there?

I would also make sure the boatloader is up to date first - do a DFU flash of the entire firmware or update the bootloader from the firmware.

arkarkark, 2025-01-28:
I formatted the drive using my ubuntu system and put back the sdcard contents and have been playing around with it tonight and haven't got it to corrupt any files. I did notice that some of my models are unix newline only files and some were dos crlf files. But I went through each of them and hit the trim button and now they're all crlf (apart from the readme.txt which was still unix style)

I'm not flying for a couple of weeks, but I'll report back if I have more problems with it.

arkarkark, 2025-01-28:
spoke too soon. I was playing around with the ELRS script lua and my 'dronesim' model just got cleared out :(

arkarkark, 2025-01-28:
and to be clear, I only ever unplug the radio from usb AFTER it has been ejected (the fs is unmounted).

arkarkark, 2025-01-28:
Alright I can reproduce this constantly now and I'm wondering if it's related to timers? I've attached my simdrone model 23 and it has a flight timer that's always increasing. every time I plug in the USB this file gets corrupted down to **exactly** 1K I've attached the corrupted file as well. notice how it's 1024 bytes long. each corrupted file is the same length and the only difference between them is the flight timer value.

[model23.yml.txt](https://github.com/user-attachments/files/18567808/model23.yml.txt)
[broken.model23.yml.txt](https://github.com/user-attachments/files/18567811/broken.model23.yml.txt)

now this might not be related to the original bug (GX12 hangs during flight) this might be a new bug, model file corrupted when plugging in USB?

philmoz, 2025-01-28:
> I just selected a model with a timer, but the timer was not incrementing and plugged in the USB and _that_ model got corrupted. so it seems only the currently selected model is corrupted when you plug in the USB.

Did you do anything after plugging into USB (e.g. copy files)?
I've tried on two different Macs (Intel with USB-A to USB-C cable, and M3 with USB-C to USB-C cable).
Not seeing any corruption.

arkarkark, 2025-01-28:
I plug in the usb and `ls /volumes/gx12/models/` and see the model file of the currently selected model is 1k in size. all other models look o.k. (size wise)

philmoz, 2025-01-28:
When you connect the USB, the firmware saves any pending changes to the model file, then shutdown the SD card so it can be connected as a drive on the USB port. 

If a directory listing is showing a truncated file then there is something very strange going on - I can't reproduce this on my GX12.

Can you try with a new blank model that has no timers, scripts or module enabled.

MEBeans, 2025-01-29:
Just like Arkarkark discovered, I too just discovered that it is something about plugging in the usb that is messing with my models. This fault also changes the channel order. Also learned that pressing one of the menu navigation keys advances the screen through whatever faults it is experiencing when you end up with a red power button, then puts the radio into calibration mode. So far, any radios that I have selected to backup on the "SD" seem to be stable on the sd, in spite of the usb doing something to their names. This fault changes the names to "MODELXX". Once the model is reloaded, the name is restored.
I'm sorry if my newness to all this is missing the mark for devs who might be looking into this, but I am just doing exactly what researching EdgeTX said you do when having problems. I am not a programmer and I have almost no experience with navigating this stuff, but I assure you, I can learn!

arkarkark, 2025-01-29:
sorry @philmoz , not going to have time to do this before a trip, I'll get back to it in a week.

arkarkark, 2025-01-29:
@3djc I have mine automatically set to be usb drive mode. which is why I'm able to see that file file is only 1K in length. I think yours is the first mention of joystick mode in this issue.

arkarkark, 2025-01-29:
oof, (again!), well I have a linux laptop that I use in the field and when I get back I can exclusively connect to that to do my configuration and see if it pops up again. and I do use joystick mode with my mac but via bluetooth ELRS, I can switch to using plugged in joystick mode when I play sim drones.

I hadn't heard about OSX messing with sdcards, but I can believe it. I've found the most reliabl;e way to do sdcards is to format them from my linux machione. OSX formatted ones (still FAT fs)  sometimes cause trouble with my HDZero goggles.

I have had a Jumper T-Pro and T-Pro v2 for years and never had this problem with them.

3djc, 2025-01-29:
Spotlight indexing engine part of MacOSX leaves (and therefore has to write) tons of hidden files. Not suggesting that it is linked, just want to remove variables from the equation. It would be nice to see if you can replicate using joystick mode instead of storage

MEBeans, 2025-01-29:
I am curious about these "nightlys". What are these? I haven't been able to nail down a pattern with the USB problem on mine yet, things still seem to have a random quality to them. The problem affects the models as well as radio configuration. Every time it happens to me , the channel order gets rearranged.

MEBeans, 2025-01-29:
3djc, I have had my radio reset models and controls on it's own over night. I was flying my drone and flying on the sim. Turned the radio off, went to bed. Next morning the radio and model data were reset. I'm sorry if my info isn't helping much, but I figure every little bit of info pointing to the problem should help get it sorted out.

arkarkark, 2025-01-29:
I’ve previously turned off spotlight and removed the spotlight from that drive

3djc, 2025-01-29:
`sudo fsck.fat -n  /dev/sdc1
fsck.fat 4.2 (2021-01-31)
/MODELS/._model00.yml`

That ._ file doesn't come from Etx AFAIK, so unsure where it comes from. Anyhow, using joystick instead of storage will tell us a lot of what might be happening, since we are then sure nothing is modifying files on Etx back

arkarkark, 2025-01-29:
That might have been before I turned off and removed spotlight. But I still see the problem with no spotlight and a freshly formatted “sdcard”

Will try the joystick thing in a week.

pfeerick, 2025-01-30:
> I am curious about these "nightlys". What are these?

The builds of the main development branch that are built every night, available for download here => https://github.com/EdgeTX/edgetx/releases/tag/nightly, and can also be selected in EdgeTX Buddy via the Filters -> "Include pre-releases" option. 

[image]

iStitch07, 2025-02-07:
I have same problem with this radio. After connecting to macOS as storage and safety disconnect, my radio.yaml was broken every time and gx12 going to emergency mode

3djc, 2025-02-07:
Not the same issue, yours is: https://github.com/EdgeTX/edgetx/issues/5841, please contact RM for support

arkarkark, 2025-02-07:
Since the advice on #5841  is to contact RM and get replacement hardware. and this issue is likely caused by telemetry data being written to the internal sd card. I think we can close this since I've contacted RM to get replacement hardware.

If you agree, please close this issue. if there is more testing you want me to do I'm happy to. but this issue was way less repeatable for me that #5841

closed 2025-02-07 by 3djc

nhurman, 2025-03-15:
Something similar happened to me today: the sticks still worked, but the switches did not. Got a nasty surprise on landing as the quad bounced on the ground and into a nearby tree without any way for me to disarm it.

However unlike this github issue, the radio did not seem frozen. The jogwheel still worked, as did the trims.

FW  : edgetx-gx12
Vers: nightly
Git#: 5bb76120
Date: 2025-03-11 02:54:35

I was just now able to reproduce the issue, video of the radio: https://imgur.com/a/L1tHILS

3djc, 2025-03-15:
Thats very strange it did happen with that version. RM did extensive testing and could not reproduce. It did happen exactly like you show, but with version ealier than martch 5th

3djc, 2025-03-15:
@nhurman could you double check the version actually running on the radio itself ?

nhurman, 2025-03-15:
Yes, this is was the exact version. I reflashed it back to stock, but the version above was copied from the SYS > Version page on the radio. I think I got it from EdgeTX buddy nightly or cloud build, don't remember which. But I still have the exact firmware file along with an image of the sdcard at the time.

3djc, 2025-03-15:
Can you reproduce somewhat easily ?

nhurman, 2025-03-15:
Not anymore... It happened 3-4 times in a row when I was playing with the radio while posting on the issue + the earlier incident, all within a minute just flicking the SE/SB switches

nhurman, 2025-03-15:
Reading the summary of [your PR](https://github.com/EdgeTX/edgetx/pull/5949) and especially this sentence

> the system has to parse some or all of the fat tables, which on this architecture, takes a LOT of time

I wonder if there was something in the internal sdcard filesystem that tripped it. I completely wiped / reformatted it between when it repro'd and when it didn't anymore. I still took a disk image, I could try restoring it and see what happens.

3djc, 2025-03-15:
Yes, please try all you can to reproduce.

bulesz, 2025-03-20:
Hello guys! I have the exact same issue! I have got the october version of EdgeTX, which came with the radio, and V1.2 main board. (my friend got the V1.3 main board in the radio, not sure this issue is related to the version of the main board or not)
Today when landed I could not have disarmed it. :/ 
With around 15 lipos earlier it went fine, but for example a week ago it hung right after armed it and not able disarmed it (it was scary, because the bird was a powerful 7" quadcopter.... :/
A month before I had the same hanging when setting up the radio and played with the settings.

The common: I have got LOGGING feature ENABLED with my models.

I can not reproduce at all due it is totally random and no pattern between the occasions.

:/ 

I hope somehow could figure out this issue, many thanks in advance!

3djc, 2025-03-20:
Fix will be part on 2.11 RC3 or release. You can also use a nightly including the fix

bulesz, 2025-03-20:
Hej hej! Thank you for your quick and kind reply! 
I have found a version 
"GX12 ELRS 2.11.0 Factory SD Contents and Firmware	2025/03/14"
at https://www.radiomasterrc.com/pages/firmware-updates

Does this version contain the fix?

Many thanks in advance,
Bulcsu

3djc, 2025-03-20:
No, you can get night lies from: https://github.com/EdgeTX/edgetx/releases/tag/nightly

pfeerick, 2025-03-20:
Or from https://buddy.edgetx.org/#/flash?version=nightly&filters=includePrereleases

2.11-rc3 will most likely be released in the next 24-48 hours... there is only one PR left in the queue for this round.

bulesz, 2025-03-20:
Thank you guys for your amazing help!

solipsia, 2025-04-13:
I'm having this exact same issue, which is very dangerous because it prevents the quad from disarming! I'm trying to install the nightly build with the fix, but the GX12 firmware is not included in the zip file. Should I use another radio's firmware? Or how do I update my GX12?