EdgeTX/edgetx #5826: X12S - GPS on AUX2

chphoto80, 2025-01-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Yesterday I installed firmware 2.10.5 on the Horus X12S and in Hardware I set Telemetry Mirror on AUX2. I disconnected the internal GPS module (no need to use). After turning the radio off and on, GPS is set again on AUX2. I created a custom build where I disabled the Internal GPS flag, but there was no change. After radio off and on, GPS is set on AUX2.

### Expected Behavior

I expected the settings to be retained even after the transmitter was turned off.

### Steps To Reproduce

1. HW set AUX2 to Telemetry Mirror
2. Radio Turn off
3. Radio Turn on
4. HW - AUX2 set to GPS by system

### Version

2.10.5

### Transmitter

FrSky X12

### Operating System (OS)

Windows

pfeerick, 2025-01-24:
So just to confirm, you did a build for X12 with `-DINTERNAL_GPS=N` set, on the Radio Settings => Version page it does **not** list `internalgps` (like shown below), but still keeps reverting to GPS for the AUX2 setting?

[image]

pfeerick, 2025-01-24:
Ok, I see the problem - this is blindly setting it to GPS on the X12S regardless of it is present or not...

https://github.com/EdgeTX/edgetx/blob/14adf0474ccedbd935fac7d71958946680b5fdfc/radio/src/storage/storage_common.cpp#L93-L96

chphoto80, 2025-01-24:
> So just to confirm, you did a build for X12 with `-DINTERNAL_GPS=N` set, on the Radio Settings => Version page it does **not** list `internalgps` (like shown below), but still keeps reverting to GPS for the AUX2 setting?
> 
> [image]

Yes, I make Internal GPS = No - as show on my image

[image]

PR #5871 (2025-02-07): fix(x12s): only preset AUX2 to GPS if enabled
Fixes #5826

While this does fix the issue, I do wonder if it would be better to instead set this as part of the one-shot `generalDefault()` that is run if there is no settings present, and thus not also override the user setting if they *don't* want to build custom firmware.
commit d9d126f9 (2025-02-07): fix(x12s): only preset AUX2 to GPS if enabled
Fixes #5826
commit ebda62b9 (2025-02-07): fix(x12s): only preset AUX2 to GPS if enabled
Fixes #5826
closed 2025-02-07 by pfeerick