EdgeTX/edgetx #5828: Cannot select 6POS individual switches as sources for special functions like play value

3djc, 2025-01-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

6p1 to 6p6 cannot be selected as sources for function like play value (it is possible on radio with customisable switches)

### Expected Behavior

6px can be selected for play value among other things

### Steps To Reproduce

1 ensure 6POS is calibrated to have 6Px available
Try to do:
ON PlayValue 6P1 5s enabled

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

philmoz, 2025-01-24:
The value field for PlayValue is a Source not a Switch.

So you can select the 6POS source; but not the individual 6POSx switches.

Same as can select SA source; but not SA Up, Middle or Down switches.

3djc, 2025-01-24:
Yes and no, each is also an individual switch that takes -100 and 100 value no ? (Haven't checked, not near a radio)

philmoz, 2025-01-24:
No, it's six individual switch positions.

Like SA source has 3 switch positions - up, middle and down which are either on or off.
6POS source has 6 switch positions - 1 - 6.

philmoz, 2025-01-24:
Radios with customisable switches has a source for each switch and the groups simulate how the 6POS works.

3djc, 2025-01-25:
Given 6POS is now more an history thing than anything else, and that no end users complain, I guess its not bad if it doesn't match 100% my initial idea when introduced !

closed 2025-01-25 by 3djc

pfeerick, 2025-01-25:
I guess at the end of the day it is more a question of what is it are you trying to achieve there... i.e. why would you want to play the "value" of a switch - hence probably why you can't do it??? Value of 6POS makes more sense, but I guess in this case you would like to be reporting which button is pressed, rather than 100, 60, etc.