EdgeTX/edgetx #5829: Launching (unsupported) lua scripts erases (corrupts) active model and controller settings

konstcranky, 2025-01-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I tried running the betaflight-tx-lua scripts on a Radiomaster GX12 controller.
I used them on 2.10 without any problems.

Now I get a lua error when trying to run the scripts, and the controller and current model settings are also corrupted. Checking the storage under Windows finds errors in the storage (lost files).

I believe that the script error occurs due to the transition in edgetx 2.11 to the new lua interpreter.
I'm ok with that, but in any case, this should not lead to corruption of system files.

I have several scripts in the repository that terminate with an error (and which work fine on the previous firmware version)

1. EdgeTX Quad telemetry dashboard - https://github.com/mvaldesshc/advanced-edgetx-dashboard
2. TBS Agent lite 0.98 - https://www.team-blacksheep.com/tbs-agent-lite-098.zip
3. betaflight-tx-lua-scripts - https://github.com/betaflight/betaflight-tx-lua-scripts

Attempting to run them often leads to corruption of system files.

GX12 is running this firmware from the factory
2.11.0-selfbuild .
GIT# d5552c80
2024-10-04 10:17:34

also tried on latesn nigtly build:
gx12-35e8b61.bin

With the same result.

I also searched for similar problems and found the following (not sure if this is related):
https://github.com/EdgeTX/edgetx/issues/5810
https://github.com/EdgeTX/edgetx/issues/5077
https://github.com/EdgeTX/edgetx/issues/5702

### Expected Behavior

The current hardware configuration and a model remains unchanged.

### Steps To Reproduce

Download https://github.com/betaflight/betaflight-tx-lua-scripts/releases/tag/1.7.0 and copy to internal storage.
Run "Betaflight setup" and "Betaflight CMS" scripts. Both should finish with error.

Turn off the controller. then try turning it on. You will most likely get an "Missing or bad radio data" error.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster GX12

### Operating System (OS)

Windows

gagarinlg, 2025-01-24:
corrupting the filesystem is, obviously, not intentional

all .luac files from EdgeTY <2.11 have to be deleted when updating to 2.11
Scripts supplied only as luac, like TBS Agent Lite, have to be updated. TBS will update Agent Lite when we release 2.11
@pfeerick can you ping betaflight to check the compatibility of their scripts to EdgeTX 2.11

konstcranky, 2025-01-24:
Yes, I know about lua files, and I've completely deleted them after update. It doesn't change the behavior.

gagarinlg, 2025-01-24:
I just installed the Betaflight scripts. They did run without issues on my GX12, but I did not have any model connected

konstcranky, 2025-01-24:
> I just installed the Betaflight scripts. They did run without issues on my GX12, but I did not have any model connected

Please tell me what edgetx firmware version you are using, and have you run betaflight scripts version 1.7.0?

konstcranky, 2025-01-24:
> Current Nightly on the GX12, installed via buddy and 1.7.0 of the betaflight scripts

can you try to install https://github.com/mvaldesshc/advanced-edgetx-dashboard ?
installation procedure is describe on their github page.

I have an error while launching Betaflight LUA now - "Syntax error - luaLoadScripts(standa) The script did not return a table

pfeerick, 2025-01-24:
Make sure you are running the latest nightly (or one within the last
day)... There was possibly a couple of nighties just recently for b&w
where entering the tools menu (which you could thus easily confuse for
running a Lua) would scramble memory and thus corrupt model settings.
Pretty sure that is gone now (but obviously aftereffects remain as memory
was saved to disk).

konstcranky, 2025-01-24:
> Make sure you are running the latest nightly (or one within the last
> day)... There was possibly a couple of nighties just recently for b&w
> where entering the tools menu (which you could thus easily [...]

Thank you for your explanation, [pfeerick](https://github.com/pfeerick).
I'm sure I tested on the two versions I indicated in the problem description, but let me check again so I don't mislead you.

konstcranky, 2025-01-25:
The problem does not occur in the latest nightly build.
Thanks again @pfeerick for the explanation. I'm most likely encountered this exact problem. I think this issue should be closed?

pfeerick, 2025-01-25:
I believe so, yes... it was just really bad timing as a couple days either way and you wouldn't have even noticed/encountered it 😆 Thanks for the confirmation...

closed 2025-01-25 by pfeerick

hawku, 2025-01-31:
I've got this problem with the latest nightly (Radio version page says GIT#: 21012d31, flashed with Buddy through DFU)
Tried to compile betaflight lua scripts and it corrupted radio.yml, model00.yml and at least common.lua.
Radio wrote some random yml data to betaflight common.lua

From SCRIPTS\BF\CMS\common.lua:
`
local function cRleDecode(buf)
    local dest = {}
    local rpt = false
    local c = nil
    for i = 1, #buf do
        if (rpt == false) then
            c = bit32.band(buf[i], CONST.bitmask.rleCharValueMask)
            if (bit32.band(buf[i], tleWarning: 0
disableTelemetryWarning: 0
showInstanceIds: 0
checklistInteractive: 0
customThrottleWarningPosition: 0
beepANACenter: 0
mixData: 
 -
   destCh: 0
   srcRaw: "I0"
   carryTrim: 0
   mixWarn: 0
   mltpx: ADD
   delayPrec: 0
   speedPrec: 0
   flightModes: 000000000
   weight: 100
   offset: 0
   swtch: "NONE"
   delayUp: 0
   delayDown: 0
   speedUp: 0
   speedDown: 0
   name: ""
 -
   destCh: 1
   srcRaw: "I1"
   carryTrim: 0
   mixWarn: 0
   mltpx: ADD
   delayPrence = 0
    end,
    draw = function()
        lcd.clear()
        lcd.drawText(screen.config.refresh.left, screen.config.refresh.top, screen.config.refresh.text, screen.config.textSize)
        for char = 1, #screen.buffer do
            if (screen.buffer[char] ~= 32) then -- skip spaces to avoid CPU spikes
                local c = string.char(screen.buffer[char])
                local row = math.ceil(char / screen.config.cols)
                local col = char - ((row - 1) * screen.config.cols)
                local xPos = ((col - 1) * screen.config.pixelsPerChar) + screen.config.xIndent + 1
                local yPos = ((row - 1) * screen.config.pixelsPerRow) + screen.config.yOffset + 1
                lcd.drawText(xPos, yPos, c, screen.config.textSize)
            end
        end
    end,
}
`