EdgeTX/edgetx #5839: MT12@Companion: no external module and no SBUS trainer

Tiefflieger68, 2025-01-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

selected radio in profile: Radiomaster MT12
1: 
no external module selectable. The only choice for protocol is "off"
Works well in the Radio. If external module is activ in the radio and settings are loaded into companion, an error message appears

[image]

2:
"Trainer/SBUS" is not selectable. 
Works well at the radio. If "Trainer/SBUS" is activ in the radio and settings are loaded into companion, an empty field "Trainer Mode" is shown in Companion

### Expected Behavior

1: several protocols for external module should be selectable. Eg PPM, Multi, CSFR...

2: "Trainer/SBUS" should be selectable. 

### Steps To Reproduce

1: 
run companion
select "MT12" in profile settings
create a new model
edit this model an try to enter a protocol for external module. Only "off" will provided

2:
run companion
select "MT12" in profile settings
create a new model
edit this model an try to select "Master/SBUS" as Trainer Mode

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

Windows

### OS Version

Nightly of 24th Jan 2025

pfeerick, 2025-01-27:
> 1: several protocols for external module should be selectable. Eg PPM, Multi, CRSF...

Running a build of Companion from about 4 days ago, I can select every one of these protocols without any error... however I do note PPM is missing from the list, so will look into why that is. 

[image]

> 2: "Trainer/SBUS" should be selectable.

Both Master/SBUS and Master/CPPM look to be missing.

philmoz, 2025-01-27:
If you can't select an external module type in the model settings, then the external module size is probably set to None in the radio profile settings.

[image]

pfeerick, 2025-01-27:
That also partly explains why PPM is missing, as it's not listed as valid for small external modules, when in fact it should be available

Tiefflieger68, 2025-01-27:
> If you can't select an external module type in the model settings, then the external module size is probably set to None in the radio profile settings.
> <img alt="Image" width="595" src="https://private-user-images.githubusercontent.com/9474356/406777169-0cbe00fa-6003-4d91-bf81-daf7a662adb9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzc5OTY2MTYsIm5iZiI6MTczNzk5NjMxNiwicGF0aCI6Ii85NDc0MzU2LzQwNjc3NzE2OS0wY2JlMDBmYS02MDAzLTRkOTEtYmY4MS1kYWY3YTY2MmFkYjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMjdUMTY0NTE2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzkxNDMyYzhkZTFlNTk5MDNlOGUxYWZkN2EwOTlhMTM4NGIwZDkyN2NiMGExZWQyYzlmM2E1NDQyOWM0NzQwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.8WXdGy-exI8RBClq3I9Li9WiCGvGRrJwO1VUnntagX0">

Oh - I've never took care about this feature in the past, because it is set correctly by default. Why ever it was set to "none" at my profile. In the latest Nighly it was set korretly in new profiles. May be there was an issue in earlier versions or was accidently fliped by me

pfeerick, 2025-01-27:
> thanks for referring to #5843 <https://github.com/EdgeTX/edgetx/pull/5843>
>
> should I close this issue?

Issue should automatically be closed when PR is merged (and thus issue
resolved).

PR #5843 (2025-01-28): fix(cpn): missing sbus trainer for some handsets
Fixes #5839

Summary of changes:
- some handsets were missing sbus trainer capability
- bumblebee was missing default nano module state