EdgeTX/edgetx #5841: Inputs and mixes clear themselves out. GX-12

MEBeans, 2025-01-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

This is my first EdgeTX radio. It's the GX-12.  Out of the box was fine. I flew my Air75 with it using the "fpv drone" model that shipped in the radio and all was well. One of the first things I did was to change the name of the fpv drone model model to suit the Air75. I don't know if doing that caused the inputs and mixes for that model to get erased?! This is the problem I am now up against. I have gone through the process of reconfiguring the inputs and mixes for this drone. I also set up a model for use with my simulator. The inputs and mixes have not been staying put, they clear themselves out. I have done the reconfigure process a couple of times. Last night I was flying in the simulator and flying the Air75. Everything was fine. I went to bed. I got up this morning to find that the inputs and mixes for these two models have cleared themselves out again! The two factory models that I have not touched are fine, nothing has changed with them. This all raises the question what am I missing? What am I doing wrong?

### Expected Behavior

Expected behavior is to be able to use my radio whenever I want to and not have to keep reconfiguring the inputs and mixes.

### Steps To Reproduce

Not sure how to reproduce this. As mentioned in the current behavior statement, it seems to be doing it by itself. Is there something I am not doing correctly when creating models? I followed a tutorial by Oscar Liang to learn how to create the models when I first found the inputs and mixes erased. I have created a model with all the inputs and mixes and gone back to change the name to see if changing the name created an issue. It didn't at that time.

### Version

Other (Please specify below)

### Transmitter

RadioMaster GX12

### Anything else?

The EdgeTX version is 2.11.0. I have upgraded the ELRS firmware to 3.5.3, original was 3.5.1 or 3.5.2.

raphaelcoeffic, 2025-01-26:
At best we would need a copy of `/RADIO` and `/MODELS` on the SD card to be able to help you.

MEBeans, 2025-01-26:
My system is not letting me copy the files. The folder moves but it doesn't show any contents.

MEBeans, 2025-01-26:
[README.txt](https://github.com/user-attachments/files/18550886/README.txt)

MEBeans, 2025-01-26:
I am getting a .yml file type not allowed.

MEBeans, 2025-01-26:
When I attempt backing up the model, it says SD error in spite of the SD being there and readable.

philmoz, 2025-01-26:
Mount the radio storage via USB and run a disk check on it to see if there are any errors.

MEBeans, 2025-01-26:
I have replied to the email this thread sent to me with a pic of the radio screen. It appears to have bricked itself.

MEBeans, 2025-01-26:
[image]

arkarkark, 2025-01-28:
My GX12 is also experiencing this loss of inputs and mixes, actually what happens is that my model file gets truncated to exactly 1024 bytes every time. I've reformatted my 'sdcard' and still get the same problem. it appears whatever model is currently selected is the one that gets corrupted when you plug in the USB cable. I'm using a usb-c to usb-c cable (no usb-a adapters) plugged into an m1pro macbook. I always eject the gx12 before unplugging.
I have some comments at the bottom of this bug report https://github.com/EdgeTX/edgetx/issues/5810#issuecomment-2617789101 which may be pertinent to this issue.

MEBeans, 2025-01-28:
I kept trying to fix my radio. It looks like normal operation has resumed. I pressed the reset button in the battery compartment. It did not seem to be doing anything other than turning the radio off. Subsequent attempts to turn it on were still resulting in the same outcome of "bad radio data". I looked away from the radio in frustration for a moment. I felt my left thumb touching what felt like the tele button. I looked back at the display and it said stick calibration. I performed that calibration and the radio resumed operating. This morning, my inputs and mixes were still present. Not knowing how I got here is driving me crazy!
Also.... Why is it that the EdgeTX version on the radio is 2.11.0, but the latest I have been able to find is 2.10.5?

MEBeans, 2025-01-31:
[radio.yml.txt](https://github.com/user-attachments/files/18610765/radio.yml.txt)

[radio_error.yml.txt](https://github.com/user-attachments/files/18610766/radio_error.yml.txt)

[model01.yml.txt](https://github.com/user-attachments/files/18610767/model01.yml.txt)

I have attached radio and model files requested. The model file attached is the one that I just spent a couple hours flying around on with my Air75. This profile now looks nothing like my other profiles, somehow it is not even remotely the same. My flight batteries needed charging, so I powered down for a few minutes. When I powered back on, my Air75 profile was again cleared out and as you can see, isn't even a model profile anymore. This happened without the usb involved this time. 

Here is a normal model profile:

[model02.yml.txt](https://github.com/user-attachments/files/18610797/model02.yml.txt)

MEBeans, 2025-01-31:
Another thing that is interesting is I have not modified two of the four model profiles that came installed from the factory, Delta wing and Heli-nexus. These models have had their names cleared out, but somehow recover their names on their own.

MEBeans, 2025-01-31:
The model02.yml is the Delta wing, one of the two I have done nothing with.

MEBeans, 2025-01-31:
Here's a thought; if changes are made to a model profile you need to back them up before you power the radio down. For some reason with the way this firmware is written, model profiles are not saved on the fly. This might explain why the factory profiles remain, they reside in the backup folder and for some reason a model profile relies on this backup folder and not just the profile in the models folder. Sound right?

3djc, 2025-01-31:
@MEBeans could you try to stop using USB connection and see if problem still exists when not connecting USB ?

MEBeans, 2025-01-31:
I have already verified that it does happen without the usb involved. The night I went through reconfiguring my models, flying around on both profiles that night, then getting up in the morning and finding all my profile and radio settings reset, there was no usb involved.

MEBeans, 2025-01-31:
I will concede that my inexperience must be playing a role here. The ELRS is the only thing in this radio that I have updated. For obvious reasons there are no updates for EdgeTX. The only things I have done is reconfiguration and recovery of the models, and a couple of transmitter settings that get reverted in the resets, i.e. transmitter protocol, baud rate and channel order. 
As I have mentioned in other posts, for some reason, the factory installed models, (Delta wing and Heli-nexus), seem to be 99% immune to this problem.

arkarkark, 2025-01-31:
I’ve also experienced it with no usb. I was running a nightly from about a week ago. Will confirm next week, then try a more recent nightly. I feel my radio is in a very reproducible state right now, so will collect as much data before upgrading

3djc, 2025-01-31:
Ok, I think we were under the impression that it was linked to usb, but it seems not. Sorry to appear slow, but I understand very quickly when I'm explained for a long time!

@MEBeans can you confirm you are running the firmware that your radio came with ? Possibly make a photo of the version screen.

@arkarkark can you confirm you run a much more recent nightly ?

From where I stand, if we might have introduced an issue recently, the firmware that Radiomaster uses has been quite extensively tested, so I doubt it can contain such a critical issue. That leaves:

A) the sd content it came with. @arkarkark could you delete everything on sd, and regenerate content with buddy. Create a new model(s), experiment and report ? Please do not use companion, or any coming from old sd content or another radio 

B) hardware changes or defect. Unfortunately, we won't be able to get answers from Radiomaster until feb 6th due to Chinese new year

@philmoz thoughs ?

MEBeans, 2025-01-31:
It looks like I still have the SD contents that came with the radio, I backed it up to my desktop early on.
Here's the firmware version on my radio...

[image]

It looks like a slow death here, my battery display has lost the graph component...

[image]

3djc, 2025-01-31:
The bar graph issue come from the settings of your radio, look in radio setup, adjust the scale of the bar graph, like 6.6 to 8.4

MEBeans, 2025-01-31:
Yup, it was set to 9.0 to 12.0

MEBeans, 2025-01-31:
Setting up an sd card in companion for the gx12 isn't going to work, is it?

3djc, 2025-01-31:
What are you trying to do?

MEBeans, 2025-01-31:
I just noticed there doesn't appear to be any sd card content listed for the gx12.

MEBeans, 2025-01-31:
I have been pawing through videos and available information to see if there is anything I might have missed. I was curious about the effort to replace the sd card contents.

3djc, 2025-01-31:
There is no radio specific actually. Any 128x65 radio does have the same content, so you can ask for RM boxer for example.

MEBeans, 2025-01-31:
As frustrated as I am, I still want to understand this stuff. With Spektrum, I never had to deal with any of this, everything just always worked. Some corner of my brain likes the challenge of figuring out what is going on here, and I know EdgeTX has much greater configuration potential. If I'm going to own a Radiomaster, I need to know how to work with it.

MEBeans, 2025-01-31:
An error that has existed in the Wizard loader from day one when selecting multi for aircraft type...

Thank you for all the time you spend here!! I understand this is all being done for free, so I am very grateful!

MEBeans, 2025-01-31:
[image]

philmoz, 2025-01-31:
Both failure cases (shutting down and plugging in USB for SD storage) use the same 'edgeTxClose' function.
This code saves any pending changes to the SD content, then shuts down the SD card. In normal use we do not update the SD card on every change, the SD card content is updated every 15 seconds if there is any change which is why we have the final check when shutting down.

I can think of two scenarios that might explain this:
- The final updates to the SD content are not getting written to the SD card before it is shut down. This would likely include directory table updates which could result in file truncation and file system corruption.
- At some point during normal operation, while writing to the SD card, we are corrupting the file system. This would likely not be noticed until a restart, although it would show up when switching models.

We did not see anything like this with the pre-production radios (mine is a reviewer sample).

philmoz, 2025-01-31:
Could I ask people having this problem to try this test:
- Connect the radio to your computer via USB
- Run a file system scan to check for errors on the SD card, and fix them (or re-format)
- Make sure the model and radio files are valid, replace if necessary
- Eject the SD card from your computer to ensure all data has been written.
- Turn off the radio while the USB is connected, ignore the warning and shutdown
- Restart the radio
- Make change to the current model - adjusting trim is sufficient
- Change model, make changes to this model
- Change back to the first model - check that there is no corruption and that the last change made is still there
- Repeat these steps - change a value on the current model, then change model

If you see any corruption of the models doing this then it looks like any SD updates are somehow corrupting the file system..

3djc, 2025-01-31:
Agreed, I just don't get how that would happen with factory firmware

MEBeans, 2025-01-31:
Thank you philmoz! I will run that protocol now. I just spent a couple hours removing any model files that I had previously configured, making sure all the sd content was as original, reconfiguring two model profiles, just to have the configs wiped out after shutting down, no usb involved.

philmoz, 2025-01-31:
> Agreed, I just don't get how that would happen with factory firmware

Maybe we are running the SD at too high a clock speed and we were just lucky during development.

MEBeans, 2025-01-31:
Is it normal to have models in the model folder but not be on the model list?

MEBeans, 2025-01-31:
During my last reconfig session, I made all of the configurations on both models before shutting down, all in one power on cycle of the radio. The model build was from "zero". the base controls the radio assigns when creating a model.

MEBeans, 2025-01-31:
> > Is it normal to have models in the model folder but not be on the model list?
> 
> Can happen if the file is incorrectly named - model3.yml will not work, needs to be model03.yml.

Ok, so even when a model is given a specific name, the writing process is supposed to be writing the model configs every so often into the model file under the naming convention "modelxx.yml"? The actual named file ends up in the backup file only, when backed up.

MEBeans, 2025-01-31:
> Could I ask people having this problem to try this test:
> 
> * Connect the radio to your computer via USB
> * Run a file system scan to check for errors on the SD card, and fix them (or re-format)
> * Make sure the model and radio files are valid, replace if necessary
> * Eject the SD card from your computer to ensure all data has been written.
> * Turn off the radio while the USB is connected, ignore the warning and shutdown
> * Restart the radio
> * Make change to the current model - adjusting trim is sufficient
> * Change to another model, make changes to this model
> * Change back to the first model - check that there is no corruption and that the last change made is still there
> * Repeat these steps - change a value on the current model, then change model
> 
> If you see any corruption of the models doing this then it looks like any SD updates are somehow corrupting the file system..

How do you know if a file is valid?

MEBeans, 2025-01-31:
Here's the result... Going between my two models and making changes did not cause any corruption. Powering down for a minute then back on caused one of the two to lose config. The model that was loaded when I turned the radio on, (model00) retained config after the power cycle. I selected model01 and its configs were wiped. I then went back into model00 and found its configs now wiped.

philmoz, 2025-01-31:
> How do you know if a file is valid?

You can drag them into companion. If there is a major problem it will complain. If they load you can check the settings or run the simulator.

MEBeans, 2025-01-31:
Another effect of this bug was that the channel order and mode would get changed, as well as the baudrate.

philmoz, 2025-01-31:
> Another effect of this bug was that the channel order and mode would get changed, as well as the baudrate.

If you connect to your computer are there any errors in the SD card file system?

MEBeans, 2025-01-31:
No, the scan said it was fine.

hawku, 2025-01-31:
I have similar problems: https://github.com/EdgeTX/edgetx/issues/5829#issuecomment-2628441765

I have now formatted the GX12 file system and rebuild it with latest SD card files.
EdgeTX version is latest nightly (GIT# 21012d3)

This was the file system contents without sounds before compiling the Betaflight LUA scripts:
[GX12FilesBeforeBFLUA.zip](https://github.com/user-attachments/files/18625444/GX12FilesBeforeBFLUA.zip)

Tried to compile BF lua script and got this error:
[image]

This is after trying to compile both BF scripts few times and power cycled the radio:
[GX12FilesAfterBFLUA.zip](https://github.com/user-attachments/files/18625469/GX12FilesAfterBFLUA.zip)

Model00.yml have been set to default values and radio.yml is empty.

3djc, 2025-02-01:
> > Agreed, I just don't get how that would happen with factory firmware
> 
> Maybe we are running the SD at too high a clock speed and we were just lucky during development.

I could believe that if there had not been extended testing by RM, including in temperature stress situations, and many units sent to reviewer, none ever reporting anything like this.

Also, this is also using the same code as any other BW radio, so wrong speed is rather unlikely

MEBeans, 2025-02-01:
I did do a rebuild of the sd contents with Boxer files. I did another batch of model configuring, backing up all changes made. I try to lump all my changes into one swoop so I don't have too many backup files.

hawku, 2025-02-01:
I have now managed to compile the Betaflight LUA scripts by rebuilding the file system again.
I added a 100 MiB padding file to a clean file system and then copied the other files.

Model00.yml and radio.yml still resets every time after a LUA script compile and power cycle.
Wizard.lua at first boot -> corrupt
Betaflight LUA scripts -> corrupt
Adding new board to BF lua script -> corrupt
ELRS lua script -> corrupt

Scripts seems to be working correctly after compilation, but it is not possible to compile anything without corrupting the file system.
Maybe the LUA compiler or FatFS library doesn't close the file system correctly?

3djc, 2025-02-01:
Not linked to LUA compiler, since there are now several scenarios causing corruption, not involving LUA at all. There is corruption somewhere, we are still unsure where as we cannot replicate

hawku, 2025-02-01:
> Not linked to LUA compiler, since there are now several scenarios causing corruption, not involving LUA at all. There is corruption somewhere, we are still unsure where as we cannot replicate

It seems that the corruption problem is happening with the pre-ordered radios. Are you testing with a pre-release model?
Do you know if RM have tried to replicate this problem with different mainboards revisions?

3djc, 2025-02-01:
Because of Chinese New Year, we won't have answers from Radiomaster until Feb 6th

MEBeans, 2025-02-01:
Perhaps I need to just ship my radio to you guys, then you will see what is going on in person.

gagarinlg, 2025-02-01:
where are you located?

MEBeans, 2025-02-01:
The issue does have a slightly random quality so far, but it has happened often enough to be a real inconvenience. For me, outside of this problem, the radio performance feels great.

MEBeans, 2025-02-01:
I'm in Maine, USA

gagarinlg, 2025-02-01:
Then it does not make sense to ship it. You would have to send it either to Europe or to Australia.

3djc, 2025-02-01:
Radiomaster management is aware of the issue, once the re open, they are committed to get this sorted one way or another. Right now, I guess we need to be patient.

MEBeans, 2025-02-01:
> Then it does not make sense to ship it. You would have to send it either to Europe or to Australia.

Yeah, I figured that. Probably would take too long considering the desire for answers.

MEBeans, 2025-02-01:
At least I can fly with it in between the reconfigurations.😉

MEBeans, 2025-02-01:
> [image]

Lol, I didn't yet understand that fault is essentially "normal", it wasn't bricked.

[134 more replies]

closed 2025-02-07 by 3djc