EdgeTX/edgetx #5848: Issue with ELRS Module Display After Updating FrSky X-Lite to EdgeTx Centurion v2.10.5

edilsoncorrea, 2025-01-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I recently updated my FrSky X-Lite to EdgeTx Centurion v2.10.5 (release date: 2024-10-13 09:54:55, firmware: edgetx-xlite). I am using an ELRS 2.4TX module, specifically the AION 2.4TX NANO 500mW model, which is updated to version 3.5.3. My X-Lite is also running the latest Lua script, elrsV3.lua.

When using OpenTx, the ELRS module’s display would power on normally whenever CRSF was enabled on the selected model. However, after updating to EdgeTx, while communication with receivers is working flawlessly, the module’s display no longer powers on.

The only scenario where the display turns on is when I disable CRSF on the model and re-enable it. If the X-Lite remains powered on, the display stays on and correctly shows when the module is connected to the receiver. However, if I power off the X-Lite and turn it back on, the display stops working again.

### Expected Behavior

The ELRS module’s display should power on automatically whenever CRSF is enabled on the selected model in EdgeTx, just as it did in OpenTx. The display should remain on and function normally, showing the connection status with the receiver, without requiring any additional steps like toggling CRSF on and off.

### Steps To Reproduce

Update the FrSky X-Lite to EdgeTx Centurion v2.10.5 (firmware: edgetx-xlite).
Use an ELRS 2.4TX module (e.g., AION 2.4TX NANO 500mW) updated to version 3.5.3.
Ensure the Lua script elrsV3.lua is installed and up to date on the X-Lite.
Enable CRSF on a selected model in EdgeTx.
Power on the X-Lite and check if the ELRS module’s display turns on automatically.
If the display doesn’t turn on, toggle CRSF off and then back on.
Observe that the display works normally while the X-Lite remains powered on.
Power off the X-Lite and turn it back on, then check if the ELRS display powers on again.

### Version

2.10.5

### Transmitter

FrSky X9 Lite / Lite S

### Operating System (OS)

Windows

### OS Version

Windows 10

pfeerick, 2025-01-27:
Have you checked in with the ELRS guys yet? As if the module is actually
working (i.e. you said the RXs are working flawlessly) and you can use the
Lua script, this may be some weird bug in ELRS. EdgeTX doesn't control the
screen on the module, it only cares about turning the module on or off, and
sending your control inputs to the module (for the most part). If this was
happening in conjunction with changing models (while the handset is on), I
would have suggested it is a hardware compatibility issue, as it seems some
modules have long power off times, so don't properly switch off and on when
changing models, but not sure what else could be happening here since you
state it is also happening when you initially turn on the handset.

edilsoncorrea, 2025-01-28:
I still believe there might be something related to EdgeTx, mainly because the issue with the ELRS display started right after I updated to EdgeTx. The only change I made was updating to EdgeTx, as the ELRS module was already running version 3.5.3 prior to the update.

mha1, 2025-01-28:
I think display not turning on needs to be understood as module is not working at all = not powered. Powering the module is solely under EdgeTX control.

@edilsoncorrea please try to reproduce the problem. If the module's display hasn't turned on remove the module from the radio and measure the voltage between the GND and VCC pins if you can.

3djc, 2025-01-28:
> I think display not turning on needs to be understood as module is not working at all = not powered. Powering the module is solely under EdgeTX control. [...]

Yes, but he says:
 "However, after updating to EdgeTx, while communication with receivers is working flawlessly, the module’s display no longer powers on."

which strongly suggests module in ON

mha1, 2025-01-28:
True, maybe @edilsoncorrea can confirm again that the module is indeed working, ie communicating with the receiver and the lua script is loading even if the display didn't turn on.

edilsoncorrea, 2025-01-28:
Yes, everything works normally except for the display on the module.

3djc, 2025-01-28:
@mha1, assuming module work, wouldn't arming toggle display on module?

mha1, 2025-01-28:
No, the display should be on as soon as the module is powered and stay on regardless of the arming status. I have two of the same modules and never saw anything like you describe. Have you ever messed with modifying the hardware configuration of the module (hardware.html)?

edilsoncorrea, 2025-01-28:
I will carry out the requested tests, measuring the voltage on VCC. I should note that the module is functioning normally, with the main LED operating correctly and even changing colors. However, the display does not turn on.

3djc, 2025-01-28:
Then I hardly see how Etx could be the cause at this point

mha1, 2025-01-28:
@edilsoncorrea you don't need to do that. The module is powered if the LED works and it communicates.

mha1, 2025-01-28:
@3djc I agree

edilsoncorrea, 2025-01-28:
I was wondering if the module, in addition to receiving voltage on VCC, also requires some sort of command, like an "AT Command," to turn on the display, which might not be sent by EdgeTx.

3djc, 2025-01-28:
No, there is nothing Etx does have access to that can control the display at all

mha1, 2025-01-28:
@edilsoncorrea please try to get it into the no display problem, then open the ELRS LUA script on the radio. Does it open? Does the display on the module turn on?

edilsoncorrea, 2025-01-28:
No, the display does not turn on when I open the LUA script. The only thing that makes the module's display turn on is when I go into the model configuration, turn off the external "CRSF" module by setting it to OFF, and then turn it back on.

mha1, 2025-01-28:
@3djc do you by any chance know what the FrSky X-Lite outputs on the module connector VCC? 5V?

mha1, 2025-01-28:
I can reproduce the problemr by just powering the stand alone module via the USB-C connector. No display but after 1 the green LED turns on indicating it wants to go to WiFi mode, ie the module is alive. The label on the module states "Voltage 6-8.4V". I believe 5V is not enough. And I remembered the NV14 outputs 5V. Same behavior as @edilsoncorrea describes. 

I'm sure this is a module specific behavior in conjunction with a low or weak power supply.

Specification:

Product: 2.4G ELRS TX module jumper
Model Name: AION TX NANO
Screen: OLED screen
MCU: ESP32
RF output power: 10-500mw
Frequency range: 2.4GHz-2.5GHz
Firmware: ExpressLRS
Antenna: Directivity diamond antenna for better signal input
Working voltage: DC5-8.4v
Receive Max Refresh Rate: 500Hz
Menu button: 5 direction button
Voltage: DC6-8.4V                        <------------------- 
Current: <400mA
Port: S port.
Firmware update: via Wifi, UBC-C or Blue tooth

edilsoncorrea, 2025-01-28:
Let me remind you that the module was working normally when Opentx was installed. This bug started happening when I updated to Etx.

3djc, 2025-01-28:
And I think we are trying to tell you that we (Etx and ELRS devs) think it is not possible, and are looking for something else that could explain that coincidence. What is current voltage reading on main screen?

edilsoncorrea, 2025-01-28:
And I think I'm trying to explain to you that if it worked with OpenTx and stopped working with Etx, it's not a coincidence. Etx changed the behavior of the power pin, which, if similar to the Taranis X9D+, is controlled by a FET.

mha1, 2025-01-28:
> I will carry out the requested tests, measuring the voltage on VCC

@edilsoncorrea  Change in plan, I'd be interested in a voltage measurement directly at the module connector

mha1, 2025-01-28:
sounds good

edilsoncorrea, 2025-01-28:
As @mha1 requested, I measured the GND and VMain pins. The voltage at this moment is 7.616V. What I believe has changed in Etx compared to OpenTx is that the "Heart Beat" data pin might have changed its initial value when the X-Lite powers on. For example: In OpenTx, it was high, and now in Etx, it’s low, or vice versa.

edilsoncorrea, 2025-01-28:
I'll make a cable with just VCC and GND and connect the module to the X-Lite with only that, isolating the Heart Beat pin, to see if the module powers on in this scenario.

edilsoncorrea, 2025-01-28:
https://youtube.com/shorts/MgIL_WIH9ww?si=LAdp22ymM7drFfQK

edilsoncorrea, 2025-01-28:
https://youtu.be/GYC3HCU6NTY?si=k2k5QvdbswTYPiSy

edilsoncorrea, 2025-01-28:
In the first video, I show that the display turns on when I switch between CRSF and OFF on the model. In the second video, I show that the module does power on via USB with 5V.

mha1, 2025-01-28:
> I'll make a cable with just VCC and GND and connect the module to the X-Lite with only that, isolating the Heart Beat pin, to see if the module powers on in this scenario.

The module is only using GND, VCC and the SPort pin for the communication ETX<->module.

> In the first video, I show that the display turns on when I switch between CRSF and OFF on the model. In the second video, I show that the module does power on via USB with 5V.

One of my Jumper AION Nano TX modules works at 5V, the other doesn't start with power via the USB-C connector and won't start reliably on the NV14 (5V VCC)

Anyhow, for now I can't see EdgeTX being part of this but I also can't think of an ELRS fw issue. Even the module that doesn't start up reliably on the NV14 starts and works fine on the TX16s and T15. The difference between the NV14 and the TX16s/T15 being NV14 5V and TX16s/T15 battery voltage (6.4-8.4V). So I'd still lean into thinking this might be related to some power issue.

If you happen to have a scope tracing the voltage during the power on of the module (connected to the radio) would be interesting.

edilsoncorrea, 2025-01-28:
As I had planned, I only connected the GND and VCC from the X-Lite to the ELRS module. I intentionally did not connect either the Heart Beat pin or the F.Port pin. I believe the most important pin here is the Heart Beat. The module powered on normally. Obviously, it then entered No Handset mode. I believe this supports my theory that EdgeTx changed the initial value of the Heart Beat pin to low during the X-Lite boot.

edilsoncorrea, 2025-01-28:
> The module is only using GND, VCC and the SPort pin for the communication ETX<->module.
>
>
> One of my Jumper AION Nano TX modules works at 5V, the other doesn't start with power via the USB-C [...]

I have an oscilloscope. I've already done some preliminary tests, and from what I've observed, it appears to be pure DC, but I will perform more tests. The VCC stays off until the radio finishes booting and after the axis checks are completed.
> If you happen to have a scope tracing the voltage during the power on of the module (connected to the radio) would be interesting.

edilsoncorrea, 2025-01-28:
I just took a measurement. While the X-Lite is booting and until the axis checks are completed, at the moment the VCC is powered, the F.Port pin is at a high state (3.3V). I believe EdgeTX changed the state of this pin during the transmitter's boot process. My next test will be to prepare to connect the F.Port to the module right after the X-Lite has finished booting.

edilsoncorrea, 2025-01-28:
I tested the voltage on the F.Port pin. As you can see in the attached video, the value of this pin remains high until the X-Lite has finished booting, checked the axis and switches. I believe this behavior was changed in EdgeTX compared to OpenTX's behavior.
https://youtu.be/D-hMRltBl8E?si=_at2PyVDGLk_wk08

edilsoncorrea, 2025-01-28:
I connected the oscilloscope, and it only starts showing communication signals when the X-Lite finishes booting. What I can’t understand is why it worked with OTX and stopped working after I updated to ETX. I’ll prepare to test it by letting the X-Lite finish booting and then connecting the S.Port right after because I believe the fact that the S.Port pin is high at that moment is preventing the ELRS module from booting.

mha1, 2025-01-28:
According to your previous post the module boots and runs. Otherwise you couldn't load the lua script and it wouldn't connect to a receiver

edilsoncorrea, 2025-01-28:
I believe the videos I sent you clearly show the issue. I ran the tests you requested, and I’ll do one more test as described above to see if it works. You believe the fact that the S.Port pin is high wouldn’t be the problem, but if that’s the case, why did the module power on normally when I connected the X-Lite with only the GND and VCC pins?

edilsoncorrea, 2025-01-28:
> According to your previous post the module boots and runs. Otherwise you couldn't load the lua script and it wouldn't connect to a receiver

The module boots up and runs because after I turned on the radio, I went to the model settings, switched it to OFF, and then back to CRSF (which powered the module again). At that point, the S.Port pin is already low, and the module is able to boot.

mha1, 2025-01-28:
ok, the question specifically was "does the lua script load after the first radio on (no display)". The question was NOT if the lua script loads after you turned CRSF off and on again (now display on)

edilsoncorrea, 2025-01-28:
https://www.youtube.com/shorts/N2z_DTpDkFk
https://www.youtube.com/shorts/a98xd_CFfp4

edilsoncorrea, 2025-01-28:
Here are the two videos. In the first video, the S. Port is connected from the beginning. In the second video, the S. Port pin is connected after the module is powered on. This behavior only occurs in the Etx, not in the OTX.

edilsoncorrea, 2025-01-28:
The LUA script loads even with the display turned off. The connection with the receivers also works.

> ok, the question specifically was "does the lua script load after the first radio on (no display)". The question was NOT if the lua script loads after you turned CRSF off and on again (now display on)

mha1, 2025-01-28:
Here's a LA trace of my NV14 booting and starting up the module.
3 = S.Port pin
2 = VCC pin

You can see EdgeTX starting and setting up the S.Port pin (at about 5s). At about 10s you can see EdgeTX powering the module and staring to communicate.

[image]

Now zooming in at 10s where EdgeTX powers the module and starts communicating. You can see EdgeTX setting S.Port low about 100us before applying power to the module. The module never sees the S.Port pin idling at high level.

[image]

If you have a logic analyzer or a 2 channel scope it'd be interesting to see if the X-Lite is different.

edilsoncorrea, 2025-01-28:
https://www.youtube.com/shorts/3jyDJ4_BTTI

mha1, 2025-01-28:
you need to zoom in to the point where the module gets power to really be able to judge the timing relation between SPort and VCC.

mha1, 2025-01-28:
The really interesting part is communication starts and is interrupted again after two data packets before resuming a little later again.

@3djc any idea

edilsoncorrea, 2025-01-28:
With a simple switch, I’ve already shown you what the problem is. I used an oscilloscope with two channels to show you that the S. Port pin has been high from the start, and this is preventing the display drivers from loading. I’ve already mentioned that the Otx didn’t do this. What more do you need to understand that the Etx changed this behavior?

mha1, 2025-01-28:
Trying to understand the root cause. Look at my zoomed in screenshot and you'll see SPort is 100us prior to power on off the module low. I can't resolve this in your first picture. Anyhow, I'm trying to get to the root cause. I'm sure the case is clear for you but it really isn't.

And for that reason, I'm out.

edilsoncorrea, 2025-01-28:
https://www.youtube.com/watch?v=N1OrlgIignA

edilsoncorrea, 2025-01-30:
I redid the test with a Taranis X9D+, and the module powers on normally. I connected both oscilloscope channels to the extension pins and verified that the S. Port pin stays low when the Taranis X9D+ is powered on and remains low until communication starts, right after the boot is completed.

I tried capturing with the oscilloscope on the modified X-Lite, adjusting the time for a better "zoom," but it wasn’t enough to determine if the S. Port pin goes back to a low state moments before VCC is powered on.

edilsoncorrea, 2025-02-05:
I was able to capture the states of the VCC and S.Port pins during the boot process of the FrSky X-Lite using a Logic Analyzer. I recorded the data similarly to what @mha1 did. I believe this data might help in understanding what may have changed in EdgeTx compared to the previous behavior in OpenTx.

[image]
[image]
[image]

mha1, 2025-02-05:
Picture #3 shows an idle high data stream which is different to your scope video https://www.youtube.com/watch?v=N1OrlgIignA. Does polarity change after doing the CRSF OFF and ON again?

Anyhow. if this is CRSF the module auto detects polarity.

edilsoncorrea, 2025-02-05:
This happens when CRSF is turned off and back on in the module. I noticed that the time until data starts being transmitted is much longer.

[image]

edilsoncorrea, 2025-02-05:
This image is from the boot process of the Taranis X9D+. The time it takes to start transmitting data after the module is powered on is much longer.
[image]

mha1, 2025-02-06:
> This image is from the boot process of the Taranis X9D+. The time it takes to start transmitting data after the module is powered on is much longer.

The dT between power on and data stream starting is 4ms (your measurement). The module itself is busy booting for at least 1/2s before it even starts looking at incoming CRSF data packets. Immediately after power power on of the module or 4ms after power on of the module doesn't make doesn't make any difference to the modules ability to auto-detect polarity and baud rate before actually starting to work with the incoming data-

edilsoncorrea, 2025-02-07:
It really doesn’t make a difference in the module’s ability to start processing data. I mentioned in one of my comments above that the module works normally. The problem is not that the module doesn’t work. The issue is that with the changes made in EdgeTx regarding these timings, the DISPLAY no longer turns on during the boot of the X-Lite. And this used to work fine with OpenTx.

edilsoncorrea, 2025-02-07:
I only have this module with a display, but I tested it on the Taranis X9D+, and it works. The high value on the S. Port at the moment the module is powered on is what is preventing the display from turning on. This is easy to test even outside of the X-Lite. I powered it with 8V and used a voltage divider with two 10k resistors to drop 4V on the S. Port pin, and the display didn’t turn on.

I’ll save the file later today and attach it here.

edilsoncorrea, 2025-02-07:
Here is the Hardware.json file:

[hardware.json](https://github.com/user-attachments/files/18709413/hardware.json)

mha1, 2025-02-07:
Your hardware.json looks ok.

I tried to reproduce the issue using your S.Port high theory on both of my Jumper AION nano TX modules (one on 3.5.2, the other on 3.5.3). A 2s LiFe (6.6V) for VCC and a two 10k resistor voltage divider to provide 3.3V at the S.Port pin. Both modules boot just fine and turn on the OLED display as they should. So far I can only conclude you might have module hw issue and recommend trying another module of the same type.

_Don't scratch your head about the outgoing colors. Red and GND are swapped to provide SPort, GND, VCC in the right order for the module. Outgoing red = GND, black = VCC (6.6V), yellow = S.Port (3.3V)_ 

[image]

edilsoncorrea, 2025-02-08:
If there is something wrong with my AION, then why did it work fine on OpenTx?

edilsoncorrea, 2025-02-08:
I believe it is now clear that the development team will not analyze what has changed in the behavior of the S.Port value for the X-Lite on EdgeTx compared to its behavior on OpenTx. I appreciate the help.

In my measurements using a Logic Analyzer, there is clearly not the same interval presented by @mah1 in his measurements. Thank you for all your help, @mah1.

I will give up trying to get my module's display to work. I still hope that in future updates, this bug will be fixed and it will work again AS IT DID ON OPENTX.

[15 more replies]