EdgeTX/edgetx #5852: CRSF Distinct GPS Altitude from Barometric Altiude

JimB40, 2025-01-29:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Seems that for now EdgeTX uses for both values the same sensor name "Alt" 
1. If FC is equiped with barometer and GPS is connected - "Alt" Sensor wil hold barometric altitude value
2. Same case if GPS is connected - "Alt" Sensor will hold GPS altitude value (barometric altitude is not available anymore)

Those are different values so would be good to retain both.

### Describe the solution you'd like

1. Barometric altitude uses STR_SENSOR_ALT "Alt" 
2. GPS altitude uses STR_SENSOR_GPSALT  "GAlt"

3djc, 2025-01-29:
Name don't have to be unique for telemetry sensors. In your example, are both actually sent by FC ? Because if they are, you should already get both today

JimB40, 2025-01-29:
AFAIU

GPS 0x02 frame

1. FC (ie Betaflight) encodes GPS data and sends CRSF frame 
https://github.com/betaflight/betaflight/blob/b277364b2cd79d7d7d7192c8a48603fe38638f67/src/main/telemetry/crsf.c#L259

2. EdgeTX gets this CRSF frame and decodes values
https://github.com/EdgeTX/edgetx/blob/79bde7e3b76335c59fee85419ed66e5ff4c3b2db/radio/src/telemetry/crossfire.cpp#L150

3.  EdgeTX creates sensor using as name STR_SENSOR_ALT "Alt"
https://github.com/EdgeTX/edgetx/blob/79bde7e3b76335c59fee85419ed66e5ff4c3b2db/radio/src/telemetry/crossfire.cpp#L58

same happens for Barometric Altitude & Vertical Speed 0x09 frame

1. BF Encoding
https://github.com/betaflight/betaflight/blob/b277364b2cd79d7d7d7192c8a48603fe38638f67/src/main/telemetry/crsf.c#L347

2. EdgeTX Decoding
https://github.com/EdgeTX/edgetx/blob/79bde7e3b76335c59fee85419ed66e5ff4c3b2db/radio/src/telemetry/crossfire.cpp#L155

3. EdgeTX creates sensor using as name STR_SENSOR_ALT "Alt" (the same)
https://github.com/EdgeTX/edgetx/blob/79bde7e3b76335c59fee85419ed66e5ff4c3b2db/radio/src/telemetry/crossfire.cpp#L65

What do you mean?
I should have two sensors named "Alt"?

3djc, 2025-01-29:
Exactly

JimB40, 2025-01-29:
0x09 frame has also Vertical Speed information. VERTICAL_SPEED_INDEX
https://github.com/EdgeTX/edgetx/blob/79bde7e3b76335c59fee85419ed66e5ff4c3b2db/radio/src/telemetry/crossfire.cpp#L172

But VSpd never shows. 
It might be on BF side that 0x09 is not send

Can we log somehow recieved CRSF frames types?

3djc, 2025-01-29:
You can mirror the telemetry stream on a serial port

closed 2026-01-22 by pfeerick