EdgeTX/edgetx #5860: CPN W64 TX12Mk2: Function of PgUp and PgDn button swapped

mha1, 2025-02-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Pressing PgDn steps from page n to n+1
Pressing PdUp steps from page n to n-1

### Expected Behavior

PgUp -> steps from page n to n+1
PgDn -> steps from page n to n-1

### Steps To Reproduce

- select radio profile TX12Mk2
- start CPN simulator
- go to model setting, observe page 1/13 (MODELSEL)
- press PgDn observe page 2/13 (SETUP)

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

Windows

### OS Version

W11

### Anything else?

Nightly:

[image]

philmoz, 2025-02-01:
That's how all radios work. It hasn't changed.

mha1, 2025-02-01:
What is "Page>"? Is it page up or down?

philmoz, 2025-02-01:
'Page>' is 'PgDn' in the new companion UI.

mha1, 2025-02-01:
On my TX16s pressing Page> advanced eg model settings to the right which equates on bw radios to page n+1. 

CPN. doesn't name the buttons Page> and Page<.   On the TX12Mk2 at least pressing PgDn results in the same manner n+1. I don't think Page> = PgDn is correct.

philmoz, 2025-02-01:
'Page>' has always been PageDn internally.

mha1, 2025-02-01:
Why don't we just name the CPN buttons Page> and Page< as the are on the real radio. Confusion removed.

[image]

elecpower, 2025-02-01:
Easy change for button text 'PageDn' to 'Page >' and 'PageUp' to 'Page <' as long as that is consistent across ALL radios.

mha1, 2025-02-02:
> Easy change for button text 'PageDn' to 'Page >' and 'PageUp' to 'Page <' as long as that is consistent across ALL radios.

Well, it's totally inconsistent now. None of the existing real radios have the buttons labelled PgUp / PgDn. And the two button radios I know (TX16s, V16, GX12) have the buttons labelled Page> and Page<

Let's do it.

elecpower, 2025-02-02:
Further investigation by @philmoz has revealed the button text is also used in radio diagnostics page so a more involved fix is required

PR #5865 (2025-02-10): fix(chore): key labels match physical radios
Fixes #5860 

Summary of changes:
- define override key labels per groups of radio targets to more closely match physical radios
- affects radio diagnostics and libsim buttons

TODO:
- add radio specific keys for all radios to lookup matrix
closed 2025-02-10 by pfeerick