EdgeTX/edgetx #5870: Frsky X9D Plus ACCTS "No SD Card" after upgrade from 2.9.3 to 2.10.6

uragano47812, 2025-02-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After the usual firmware update when i power on the TX it show "No SD card" message. So i cancel the SD and re-installed the SD content from Buddy app. Again the same error. 
I then bought a NEW SD Card 32Gb. It is FAT32 and i used Buddy to download SD content but again with no luck. 
I tried to go one step back and flashing 2.10.5 firmware but the error is still there.
 

### Expected Behavior

Work

### Steps To Reproduce

Switch TX ON

### Version

2.10.6

### Transmitter

FrSky X9D+

### Operating System (OS)

Windows

### OS Version

Win 10

DongkwanYi, 2025-02-07:
Try this tool when formatting your SD card.
https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-windows-download/

ParkerEde, 2025-02-07:
Try a 4 GB SD card. If I remember correctly, there are problems with cards larger than 16 GB on older Frsky radios.

3djc, 2025-02-07:
Nop, the issue is Windows cannot natively format FAT32 on 32GB+ SD and some users ends up with exFAT formatted SD Card which don't work. Properly formatted even large SD work fine in old FrSky gears (not fakes SD of course)

uragano47812, 2025-02-07:
The SD card that was working good on 2.9.3 is a Kingston 2GB. The new one is A NETAC CLASS 10 32Gb. As i said the new card is in FAT32 anyway i will try the tool for formatting again.

uragano47812, 2025-02-07:
Ok i have formatted the SD with the tool and downloaded the SD content again . No luck still the same problem. I will revert back to 2.9.3 and wait for a solution.

chofchop, 2025-02-10:
This issue is discussed in #4530. After #3957, some brands of SD cards are no longer recognized by EdgeTX.
Developers say they are taking this step to solve a more serious problem.