EdgeTX/edgetx #5873: Lua script : slow up/dw precision not preserved with model.insertMix

Gjeremie, 2025-02-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When we copy a mixer line and insert it as a new line WITH A LUA SCRIPT (telemetry script) the precision 0.00 of the  slow up/dn is resetted to 0.0 in the new mixer line.

### Expected Behavior

the precision slow up/dn precision of 0.00 must be  preserved

### Steps To Reproduce

1- in edgetx the mixer line for slow up/dn precision is setted to  0.00

2- add a new line to the channel with a lua script
tab = model.getMix(19, 1) -- copy the first mixer line into the Table  "tab"
model.insertMix(19, 2, tab) -- Insert the mixer line ("tab") in the second line of the channel

3- in edgetx open the mixer line : the  slow up/dn precision is resetted to  0.0

### Version

2.10.5

### Transmitter

RadioMaster MT12

pfeerick, 2025-02-07:
Looks like for starters `luaModelGetMix()` doesn't pull in the value of `speedPrec` and then `luaModelInsertMix()` doesn't write it.

pfeerick, 2025-02-07:
The linked PR appears to resolve this, and will be part of 2.11 and also pushed to 2.10 so it is present in the next update forthat branch. Just some notes / clarification regarding the snippets of code provided: 

```lua
tab = model.getMix(19, 1) -- copy the first mixer line into the Table  "tab"
```

Counting starts from 0 (i.e. CH1 = 0), not 1. So this line reads CH20, line 2. 

```lua
model.insertMix(19, 2, tab) -- Insert the mixer line ("tab") in the second line of the channel
```

Here, we are inserting into CH20, line 3.

Gjeremie, 2025-02-08:
Thanks for the fix.

Yes I made a mistake in the comment for the line number.

pfeerick, 2025-02-08:
Thanks, yes, the same problem exists for main (2.11), but not for 2.10, as delay precision is not a feature included in 2.10

closed 2025-02-08 by pfeerick

Gjeremie, 2025-02-10:
Thank you for the fix, I flashed the nightly of the 9 february 2025 and the precision 0.00 works perfect now.

BUT I have another issue , see :  #5782

PR #5874 (2025-02-08): fix(lua): missing speedPrec for getMix/insertMix
Fixes #5873 

Summary of changes:
- add missing `speedPrec` to Lua getMix() and insertMix() functions

Tested on GX12 with two mixer lines on CH20, lua function script run via SF will create the third. It beeps when it runs ;)

```lua
local tab

local function init()
  -- Called once when the script is loaded
end

local function run(event, touchState)
  -- Called periodically while the Special Function switch is on
  tab = model.getMix(19, 1) -- copy the first mixer line into the Table  "tab"
  model.insertMix(19, 2, tab) -- Insert the mixer lin [... cut]