EdgeTX/edgetx #5875: NV14 2.10.6 - no aux com port option in hardware menu

rams78, 2025-02-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There is no options to setup aux serial port, ergo not possible to use any external modules.
It used to work with OpenTX, and module (TBS Crossfire nano) surely getting power.

[image]
[image]
[image]

### Expected Behavior

[image]

### Steps To Reproduce

1. Flash latest firmware (same with older versions)
2. Open hardware menu - no aux com port options
3. Also, latest TBS companion can not find the module

### Version

2.10.6

### Transmitter

Flysky NV14

### Operating System (OS)

Windows

### OS Version

11

pfeerick, 2025-02-07:
No, that is completely incorrect usage of the AUX functionality. Go into Model Settings -> External RF and set that to CRSF. Does that work?

The AUX -> External module setting is *only* for allowing certain radios to be modded to work with FrSky ACCESS modules that need a duplex UART... if I remember correctly, NV14 doesn't even need that mod to work with them.

rams78, 2025-02-07:
Thank you for clarification. Maybe the documentation deserves a few lines to outline that those menu items are not available on all devices?
Anyway, It is possible to select External RF and CRSF protocol. However, latest TBS Agent Lite has another issue - it literally takes several minutes to react to any input, first I even thought it's just freezing, and only accidentally noticed it's actually doing something.
I guess this issue can be closed. I haven't searched for TBS agent issue yet, maybe there is a known solution.

richardclli, 2025-02-08:
> Thank you for clarification. Maybe the documentation deserves a few lines to outline that those menu items are not available on all devices? Anyway, It is possible to select External RF and CRSF [...]

The TBS Agent symptom is a hardware issue though, either do a DIY mod to remove a cap, or buy the upgrade version EL18 instead.

The following image shows which cap need to be removed
[image]

pfeerick, 2025-02-08:
There is already this text at the top of the section on serial ports ;)
Perhaps the word ports in the second sentence could be changed to "ports
and options".

> The listed ports are based on the ports that are available in the
particular radio hardware. The ports listed below are for example only and
may not be present in your radio.

https://manual.edgetx.org/color-radios/radio-settings/hardware

rams78, 2025-02-14:
@richardclli , I confirm that after removing the capacitor as you advised crossfire agent is now working smoothly! Thank you very much, you made my day!

closed 2025-02-14 by rams78