EdgeTX/edgetx #5877: Taking a screenshot

shiron007, 2025-02-07:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Offers a missing option that I think will help all users

### Describe the solution you'd like

Setting the screenshot button in advance as default.
I know there is an option to set a button for taking a screenshot. I configured it myself, and it works well, but I think it would be more convenient for everyone if the screenshot feature was always pre-configured in the system by default. Ideally, it should be activated by pressing both "T" buttons simultaneously rather than being assigned to one of the Switch buttons.

pfeerick, 2025-02-08:
Both "T" buttons?

EdgeTX runs on a multitude of different radios, with different hardware
layouts, input options, display types... I'm afraid you will have to be
more specific ;)

And I for one am generally opposed to this... screenshots are not something
I thing the general user would want to do, so I don't see the benefit in
pre-defining something not used by the majority, when it is just a matter
of configuring either a general or special function to do it if you so
wish.

Easier configuration of said SF/GF via the presets Lua tool is definitely
something that could be done though.

shiron007, 2025-02-08:
I understand, thank you very much for your attention.

closed 2025-02-16 by shiron007