EdgeTX/edgetx #5878: The lower-left telemetry button.

shiron007, 2025-02-07:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Suggests a fix

### Describe the solution you'd like

The lower-left telemetry button
The telemetry button currently takes us to the "Screen Settings" menu—why?
I suggest that a long press on it should take us to the telemetry screen, while a quick double press should take us to the screen settings.

pfeerick, 2025-02-08:
Right from day one it has always been screen layout settings, and then a
long press handler was added which also does the channel monitor.  It has
been that way for a while now, and it is probably not fair to change it on
all users now. Better to wait for key shortcuts to be customisable. Also,
there are no (and I don't think there will be any double press actions,
only long and short press).

---------- Forwarded message ---------

shiron007, 2025-02-08:
OK , I understand, thank you very much.

closed 2025-02-16 by shiron007