EdgeTX/edgetx #5880: Image preview file selection

shiron007, 2025-02-07:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I propose a solution that will help all users.

### Describe the solution you'd like

On the file selection screen from the list of image files,
I suggest having a preview of the file when hovering over it before selecting.

Gromulin, 2025-04-17:
Can I expand on this request? I would like the ability to open and display image files from the SD card onto the the screen (TX16S). Usage examples - An image of the gimbal adjustment screw locations. Displaying screen shots, etc.

pfeerick, 2025-04-17:
That is slightly different I think. This request seems to be in reference to the file/image picker for the model image, i.e. in model settings. I do agree though, it would be nice to be able to view an image full screen in radio settings -> SD card browser.