EdgeTX/edgetx #5884: Display glitch when using Lua script in mixers

agnauck, 2025-02-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using Lua mixer scripts as mixer source there is a display issue in the mixer line when the script name and output value is longer than the reserved column width

See the screenshot here on CH7:
[image]

### Expected Behavior

The script name + output value should be truncated or trimmed so that it does not exceed the reserved column width

### Steps To Reproduce

1. use a simple Lua mixer script as a mixer source.
2. when the script name and the name of the output value exceed 8 characters you should be able to reproduce the ui glitch

### Version

2.10.6

### Transmitter

RadioMaster TX16S / TX16SMK2

philmoz, 2025-02-08:
Already fixed in 2.11.

agnauck, 2025-02-15:
I tested this in 2.11 rc1.
The text is not overlapping into the next column anymore.

But when no name is specified for the custom script then it shown only icon+script-number+dash.
It looks like its truncated after the dash.

See here on channel 7:
[image]

[image]

Here a name is supplied and it shows all characters that fit into the column:
[image]

[image]

PR #5916 (2025-02-15): fix(color): mix line source string may be truncated if it is too long.
Fixes #5884 

Reduce font size if mix source string is long.
closed 2025-02-15 by pfeerick