EdgeTX/edgetx #5894: Jerky screen when scrolling across main views (touch screen).  OK when using up and down keys.

Kevltan, 2025-02-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Jerky screen when scrolling across main views (touch screen) ok when using up and down keys in 2.11.0- RC ok in 2.10.6.

### Expected Behavior

Smooth screen transitions when touch sceen scrolling through main view pages.

### Steps To Reproduce

As above.

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

Kevltan, 2025-02-11:
> Please upload the affected model file and any widgets and Lua scripts
> being used. Scrolling seems ok on my TX16S.

Will do!

Kevltan, 2025-02-12:
@philmoz
[model22.zip](https://github.com/user-attachments/files/18763674/model22.zip)

Kevltan, 2025-02-12:
@philmoz 
Hi Phil, it seems to do it even without the receiver connected. I don't know if this a coincidence but when I plug my USB cable whilst connected to the PC, the radio switches of instantly every time (I have opened another ticket for this). In 2.10.6 it does not do this and scrolling is relatively smooth. I have moved backwards and forwards through the 2 firmwares and the same always happens!

Kevltan, 2025-02-12:
@philmoz 
I have loaded a basic SD card for 2.10.6 no extra files added and as soon as the usb lead inserted the radio crashes to off. So it doesnt seem to be related to SD card content.

philmoz, 2025-02-12:
Can you upload short videos of both 2.10 and 2.11 showing the difference.

philmoz, 2025-02-13:
@Kevltan can you please post videos showing the difference between 2.10 and 2.11.

Kevltan, 2025-02-13:
Yes, I will do as soon as I can Phil👍

Kevltan, 2025-02-14:
https://github.com/user-attachments/assets/17a6e024-e2ed-4c98-943c-6f7351ca8443

2.10.6 firmware

https://github.com/user-attachments/assets/bfd419b1-7272-4dad-87c7-06b802874dc0

2.11.0 firmware

philmoz, 2025-02-14:
Does it make a difference if you remove the model image?

Kevltan, 2025-02-14:
No probs Phil I will have a look for you later 🙂😴

philmoz, 2025-02-14:
I suspect the model image is why you are seeing slightly slower scrolling.

In 2.10 and earlier the model image is scaled to the widget size using a very simple algorithm.
When the widget changes size (e.g. selecting full screen or turning off top bar or sliders), the image has to be reloaded and rescaled.

In 2.11 the image is loaded full size and scaled as needed to fit using LVGL functions. The gives a better quality image (with antialiasing); but is slightly slower to render.

2.10:
[image]

2.11:
[image]

Kevltan, 2025-02-14:
@philmoz 
Hi Phil, I have removed the PNG image from the main view, and it makes no difference whatsoever. Still the sort of double take on the rollover over between the views on 2.11. I have swapped backwards and forwards through 2.10.6 and 2.11, and 2.10.6 is definitely more smooth on the view transitions, and there is no difference in having a png image present.

inventor7777, 2025-02-18:
I can confirm the same on my TX16s. I updated to 2.11 when I bought the radio so I thought it was normal behavior :0

Mine has a custom PNG set as well.

philmoz, 2025-02-20:
After more investigation, I don't think there is much I can do here.

In 2.10 the built in widgets (radio into, date, timer, value, etc) were being drawn directly into the screen buffer (bypassing Lvgl).

In 2.11 these have all been converted to use Lvgl controls in order to be compatible with the overall styling and user selected theme.

This adds extra overhead when the tile view tries to scroll the view left or right which is why it appears less smooth.

Kevltan, 2025-02-20:
Thanks for trying Phil 🙂 Will a faster H7 processor or faster graphics
processor/higher resolution screen make this better in the future, or is
this just an LVGL artifact that we will just have to live with when
scrolling main views?

inventor7777, 2025-02-20:
Maybe in the future there could be a setting like “Reduce Motion” (like on iOS) that disables the scrolling motion altogether? It’s so choppy that it’s almost not even worth having the animation. 🤷‍♂️

pfeerick, 2025-02-27:
Closing this issue, but this does not mean we won't try to improve performance more in the future if possible.

closed 2025-02-27 by pfeerick

Kevltan, 2025-02-27:
@pfeerick 
OK by me Peter, just wanted to highlight the difference between the firmwares. Hoping that in the new faster radios, that there will be more smoother page transitions.

pfeerick, 2025-02-28:
And your feedback is much appreciated! 🤗  Newer radios (or even a TX16S MK3 without a rotated display might help there... just sayin'! 😆) should be better, but I hope we can still do some nip and tuck for the older ones as time progresses.