EdgeTX/edgetx #5895: BT selection broken for BW X7 radios

Doncuppjr, 2025-02-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Click [Discover] for the BT trainer in the model config
Browse for the correct slave
Click the slave - nothing happens
Click return to leave the dialog - nothing happens
Trapped, power cycle.

### Expected Behavior

Click [Discover] for the BT trainer in the model config
Browse for the correct slave
Click the slave - slave is paired and dialog closes.

### Steps To Reproduce

Flash 2.11 on an X7 BW Radio
Press the MDL key
Goto setup
Scroll down to Trainer
Pick mode BT/Master
Select [Discover]

### Version

2.11.0-rc

### Transmitter

RadioMaster Boxer

### Anything else?

This is also true for the GX12, and any X7 BW I would imagine, possibly even more radios.

Doncuppjr, 2025-02-11:
Looks like this is possibly already tracked, but I thought I would open an official issue.

pfeerick, 2025-02-13:
I now have a Boxer and Pocket setup with BT modules, and have gotten them to link up as master and slave, using 2.10, so have something to work with now to look into this. However, I am using the ESP32 emulator project for this, but hopefully it misbehaves just as badly as the para modules since it is supposed to be emulating them. https://btwifimod.gitbook.io/

Doncuppjr, 2025-02-13:
Nice. The other piece of that puzzle for me specifically is that I am using the ysoldak head tracker on a seeed studio sense, vs buddy boxing. https://github.com/ysoldak/HeadTracker. I have a few esp32's coming in today, so I can try the btwifimod.

Doncuppjr, 2025-02-13:
I get the same behavior with the btwifimod as the official para module.

pfeerick, 2025-02-16:
Hey @philmoz ... what are your thoughts on this?

It looks like 
https://github.com/EdgeTX/edgetx/blob/f4d5bb39081e40f6c3070a497e103723a697c54c/radio/src/gui/128x64/model_setup.cpp#L1728-L1729

keeps getting called, the popup menu keeps getting MAC addresses added to it, you can scroll the menu, but it never passes control to `onBluetoothConnectMenu` when you do a (long?) RTN or ENTER. If I remove the following line, it *appears* to work, but unsure if that is breaking a load of other stuff, or if the *fault* is in how the bluetooth menu operates now...

https://github.com/EdgeTX/edgetx/blob/f4d5bb39081e40f6c3070a497e103723a697c54c/radio/src/gui/common/stdlcd/popups.cpp#L340

philmoz, 2025-02-16:
Other popup menus have all the items pre-filled and POPUP_MENU_START is only called once on key press.
This one seems to be filling the menu as devices are discovered and keeps calling POPUP_MENU_START over and over in which case the call to killAllEvents is likely messing things up.

Maybe try this:
```C
              if (popupMenuHandler != onBluetoothConnectMenu) {
                POPUP_MENU_TITLE(STR_BT_SELECT_DEVICE);
                POPUP_MENU_START(onBluetoothConnectMenu);
              }
```

I don't have any BT hardware so I can't test it.

philmoz, 2025-02-16:
Alternatively you could modify POPUP_MENU_START and move the killAllEvents inside the if block:

```C
void POPUP_MENU_START(PopupMenuHandler handler)
{
  if (handler != popupMenuHandler) {
    killAllEvents();
    AUDIO_KEY_PRESS();
    popupMenuHandler = handler;
  }
}
```

This might be better as it would handle any other places in the code that work like the BT code.

pfeerick, 2025-02-16:
Both versions appear to resolve the issue, so second version it is.

Doncuppjr, 2025-02-16:
I wonder if we can add some more time for the discovery. I do like that the window pops up with minimal delay, but I get an almost random number of devices to pair with, and not always the one that I want.

pfeerick, 2025-02-16:
I suspect the limitation is the Bluetooth module... we probably just ask it
to do a discovery scan, and it reports back what it sees within some time
frame (which will include any "smart" watches, phones, Bluetooth sensors,
etc)...(and the menu added to as more stuff is seen) ... If that is the
case that is a more a question of *if* the discovery time can be
set/extended otherwise would need to rethink the scan dialog (or simply
press RTN and discover again to trigger a new scan).

---------- Forwarded message ---------

#5895)

<https://github.com/EdgeTX/edgetx/issues/5895#issuecomment-2661584689>, or
unsubscribe

<https://github.com/EdgeTX/edgetx/issues/5895#issuecomment-2661584689>, or
unsubscribe

Doncuppjr, 2025-02-16:
I did a comparison with a Boxer that has PARA, and the dialog takes a few seconds to come up, vs a GX12 with the BTWifiMod, which comes up almost instantly. Maybe the issue lies with the BTWifiMod project. I'll talk to Cliff about it.

Doncuppjr, 2025-02-16:
How does the BTWifiMod respond in your Boxer/Pocket radios?

pfeerick, 2025-02-16:
I'm using [these ESP32 Wemos D1 Mini boards](https://templates.blakadder.com/wemos_D1_Mini_ESP32.html) and the menu and 4-5 devices showing up pretty much instantly. It would not surprise me that the BTWifiMod doesn't stay in discovery for quite as long... as it's not 100% compatible or complete... i.e. the 0.9 release notes you need to do the rescan every time (it doesn't save the pairing) and you can't use it as a trainer with a FrSky module... but it is what it is, and it's free 😆

Doncuppjr, 2025-02-16:
I think his HeadTracker project might be better. It actually has a storage partition/and a lot more development done. You can flash the firmware and then in the HeadTracker app tell the device whether it's a slave or master. Looks very promising.

Doncuppjr, 2025-02-16:
Those look cool, I went with these, https://www.amazon.com/DWEII-ESP32-C3-Development-Bluetooth-Frequency/dp/B0CW62PPZ1/ref=sr_1_5?crid=216YBYUK3N6ZS&dib=eyJ2IjoiMSJ9.ryEhFgLjDT5LrUWm8Ur1pb9vZRKcFpFFsOOgg3AsDg9Hag8K-dheavrtfTqD2DMFjAQlG-sCBWkqctTNh2Cnso4eHAM8q9nrKFMybjfOYIPNbI4jbvJT6JUyQVtFjUh2g9wYtfcWLfRsMg0NP4wrWRX1HL-IQZKaZAI419boCzUXD4-saskRF__KeETsa1V6WP73p-HG8sDMjUaK2AB8FyaBr0FBpkPfIpV_oTPQ3uhYJXu6RzRj5QPPuOi3YPAmxP7NRUl-CKrEPdfGbWssQ-AkZcNci62t0Z37eHEyXqp9gmlCUpMtcBVkqGSBZe9SfJdUZXp6UodScx-8s84XJnFXeB05HhGO_48kS93psX7X0MvOyFYFaoV4KXB1okpN.i7ntUmSxhPmk-GKVAdI1uZSKUzBVNzfMW6E1MK0ZTRo&dib_tag=se&keywords=c3%2Bsuper%2Bmini&qid=1739747543&sprefix=c3%2Bsuper%2Bmini%2Caps%2C133&sr=8-5&th=1

closed 2025-02-16 by pfeerick

pfeerick, 2025-02-16:
Nice. Since they're S3 based you also get the possibility of some benefits from BT5. Those boards are nice, have been thinking about getting something similar but have a bunch of the [Seeed Xiao](https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html) boards so will probably try one of those for something a little more permanent, and with an external antenna for the BT. 

Yes, I think the Headtracker continuted to get a little more love simply due to more interest from others... idealy the packet based mechanism he was working on for BTWifiMod would be the way to go, as that would enable a *lot* more capability, but that is a project in of itself... 🤯 ... so it is no surprise it stalled. Hopefully it will start moving again someday :)

Doncuppjr, 2025-02-16:
I’m cheerleading for it, things are looking good.

PR #5918 (2025-02-16): fix(bw): bluetooth discovery popup menu lockup
Fixes #5895

Summary of changes:
- ensures `killAllEvents()` is only run on `POPUP_MENU_START()` starting a new menu