EdgeTX/edgetx #5897: SPI Audio has intermittent sound on trims operation

richardclli, 2025-02-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using trims, the "di di" trim sound is intermittent.  Only happens in SPI audio, DAC audio are all good.

### Expected Behavior

When using trims, the "di di" trim sound should sound properly.

### Steps To Reproduce

1. Use trim hats mode
2. Push trim
3. Sometimes has sound, sometimes no sound

### Version

2.11.0-rc

### Transmitter

Flysky NV14

pfeerick, 2025-02-13:
Does it have any connection to the audio mute setting in hardware settings? I wonder if the NV18/EL18 actually honours that setting, as I can usually hear a hiss even on a good radio, and I can't hear anything from the NV14 when not playing sound 🤔

richardclli, 2025-02-13:
> Does it have any connection to the audio mute setting in hardware settings? I wonder if the NV18/EL18 actually honours that setting, as I can usually hear a hiss even on a good radio, and I can't hear anything from the NV14 when not playing sound 🤔

Suspect some changes due to muting or SPI cause this problem.  Spot strange hardcoded SPI clock speed definitions, the change of the source of mute timing.  However, still cannot figure out a way to make it behaves as good as pre-H7 version.

richardclli, 2025-02-13:
I just tried to back port the old driver (pre-H7-base)  to main and it seems to works properly.  So, what is wrong about the new driver....... need to figure out.

richardclli, 2025-02-13:
I caught the bug finally, wrong mute logic!

PR #5905 (2025-02-13): fix(audio): clipped SPI audio on nv14/el18/x12s

Fixes #5897
closed 2025-02-13 by pfeerick